oxidecomputer/hubris #275: "getting OpenOCD" link is broken

TommyMurphyTM1234, 2021-11-30:
On this page

https://github.com/oxidecomputer/hubris

The "getting OpenOCD" link is broken and generates a 404 error.

The link is to

http://openocd.org/getting-openocd/

But should be to

https://openocd.org/pages/getting-openocd.html

I think that the OpenOCD site/pages were reorganized recently which probably caused this mismatch.

closed 2021-12-08 by kc8apf

TommyMurphyTM1234, 2021-12-09:
There another link to OpenOCD that should be going to the "getting OpenOCD" page but actually goes to the binary downloads page. Perhaps that needs to be fixed too?

https://github.com/oxidecomputer/hubris#prereqs

> If you need to build from source, you can find the OpenOCD v0.11.0 here.

jeffallen, 2021-12-14:
@TommyMurphyTM1234 No, that part of the document is encouraging users to install version 0.11.0 from source instead of older versions, so it is correctly pointing to the source archives for precisely version 0.11.0.

TommyMurphyTM1234, 2021-12-14:
Ah, thanks @jeffallen - I mistakenly thought that it was linking to binary builds and not source tarballs. My mistake! :-)

PR #309 (2021-12-08): Fix OpenOCD link
Fixes #275.
commit a52b52ed (2021-12-08): Fix OpenOCD link

Fixes #275.