oxidecomputer/hubris #281: [Stream] Need a copy-paste setup for Ubuntu

riking, 2021-12-01:
libudev-dev was missing from the instructions; create an easy copy-paste to install all system dependencies.

cbiffle, 2021-12-03:
I'm surprised we have a dependency on libudev. @bcantrill do you know where this is coming from? Do we link probe-rs into xtask or something?

In any case, we should have a list of the libraries people need to install (or a script that installs them) -- though I don't use Ubuntu so I'm not sure what they'd be, and we might want to break it out of the README if we need to have blocks of text for six different distros.

cbiffle, 2021-12-03:
Ah, @luqmana and @steveklabnik explained it to me -- we need libudev indirectly because of the UART bootloader unlocking tool in one of _our_ dependencies, lpc55_support. We should add that to the readme then.

simeonmiteff, 2021-12-05:
Don't know if this belongs here, but I failed to RTFM and got stuck with the demos on my first attempt due to an out-of-date openocd (which on Ubuntu 20.04 - the current LTS, is too old). I didn't want to teach APT about specific packages from Ubuntu 21.xx I just went with this filthy hack (use at your own risk, or check signatures separately):

```console
$ wget http://mirrors.kernel.org/ubuntu/pool/universe/o/openocd/openocd_0.11.0~rc2-1_amd64.deb
$ wget http://mirrors.kernel.org/ubuntu/pool/universe/libj/libjaylink/libjaylink0_0.2.0-1_amd64.deb
$ wget http://mirrors.kernel.org/ubuntu/pool/universe/libg/libgpiod/libgpiod2_1.6.2-1_amd64.deb
$ wget http://mirrors.kernel.org/ubuntu/pool/universe/libf/libftdi1/libftdi1-2_1.5-5build1_amd64.deb
$ wget http://mirrors.kernel.org/ubuntu/pool/universe/c/capstone/libcapstone4_4.0.2-3_amd64.deb
$ sudo dpkg -i *.deb
```