oxidecomputer/hubris #284: qemu support

lu-zero, 2021-12-02:
Would be nice to support a system simulated by qemu and document how to bring it up.

cbiffle, 2021-12-03:
I'm glad you asked! I'm working on a FAQ and QEMU is part of it.

QEMU's system-level emulation of the Cortex-M interrupt model, NVIC, SysTick, and other core components is not correct enough to usefully test Hubris. This has been the case for as long as I've been trying to use it for stuff like this (circa 2011) and we re-tested this past spring and it's still true. Hubris might _run_ under QEMU, but test results wouldn't reflect actual hardware, making it less useful for us. (If I recall correctly we did experience random hangs on QEMU; your mileage may vary.)

We've considered trying to fix it, but it doesn't look easy, and we are a small team focused on getting our product ready.

cbiffle, 2021-12-03:
@bcantrill has reminded me that we've actually got a repo with one of the QEMU bug cases we hit. https://github.com/oxidecomputer/qemu-systick-bug

lu-zero, 2021-12-03:
Hopefully the current qemu might be less buggy, if you have already some guide to test it I can try to give it a go and report back :)

kc8apf, 2021-12-03:
https://xpack.github.io/qemu-arm/ claims to emulate STM32F4 Discovery.  Hubris supports that board via app/demo-stm32f4-discovery/app.toml.  Let us know how it goes.

lu-zero, 2021-12-06:
I tried to do the following:
```
xpack-qemu-arm-2.8.0-13/bin/qemu-system-gnuarmeclipse -board STM32F4-Discovery --image /Users/lu_zero/Sources/rust/hubris/target/thumbv7em-none-eabihf/release/demo-stm32f4-discovery
qemu-system-gnuarmeclipse: Attempt to set CP10/11 in SCB->CPACR, but FP is not supported yet.
```

And after that it seems hanging, having some step-by-step instructions would be nice to have ^^;

kc8apf, 2021-12-06:
As we said before, our experience is that qemu's Cortex-M models are
inaccurate enough that we don't use qemu with Hubris at Oxide.  We don't
have any step-by-step instructions because we don't know and it's not a
path that we intend to spend any time investigating.

As for the error message you got, Hubris assumes the device has hardware
floating point.  Qemu's model apparently doesn't implement it. You'd need
to add floating point support to qemu's model as stm32f4 discovery has
floating point.

cbiffle, 2021-12-13:
It's surprising that they would claim support for STM32F4 if they don't have floating point turned on. I would expect most programs written for STM32F4 these days are built with floating point support.

shua, 2022-03-18:
The open ticket listed in https://github.com/oxidecomputer/qemu-systick-bug for the arm systick issue https://bugs.launchpad.net/qemu/+bug/1872237 has been closed as fixed. I imagine plenty more still outstanding, but figured I'd update this ticket.