oxidecomputer/hubris #295: Incorrect path to gdb scripts for f3/f4 discovery demos in README.mkdn

simeonmiteff, 2021-12-05:
While trying out hubris on my stm32f3discovery I encountered this error when running `cargo xtask gdb app/demo-stm32f4-discovery/app-f3.toml openocd-f3.gdb` (it causes the demo not to run):

```console
<snip>
openocd-f3.gdb: No such file or directory.
(gdb)
```
Specifying the full path for the gdb script works for the F3 demo (and I presume the same would be true for the F4 demo):
```diff
diff --git a/README.mkdn b/README.mkdn
index d7813c9..4cccde6 100644
--- a/README.mkdn
+++ b/README.mkdn
@@ -182,8 +182,10 @@ $ openocd -f app/demo-stm32f4-discovery/openocd-f3.cfg    # STM32F3 Discovery bo
 From another terminal:
 
 ```console
-$ cargo xtask gdb app/demo-stm32f4-discovery/app.toml openocd.gdb         # STM32F4 Discovery boards OR
-$ cargo xtask gdb app/demo-stm32f4-discovery/app-f3.toml openocd-f3.gdb   # STM32F3 Discovery boards
+# STM32F4 Discovery boards
+$ cargo xtask gdb app/demo-stm32f4-discovery/app.toml app/demo-stm32f4-discovery/openocd.gdb
+# OR STM32F3 Discovery boards
+$ cargo xtask gdb app/demo-stm32f4-discovery/app-f3.toml app/demo-stm32f4-discovery/openocd-f3.gdb
 ```
 
 If this works, you'll see `gdb` mumble some things about flashing, `openocd`
```

jmingov, 2021-12-05:
Should be fixed also for the other boards or at least for the 753.

`cargo xtask gdb app/demo-stm32h7-nucleo/app-h753.toml app/demo-stm32h7-nucleo/openocd.gdb`

simeonmiteff, 2022-01-11:
Indeed - fixed by #331, thanks @jeffallen.

closed 2022-01-11 by simeonmiteff