oxidecomputer/hubris #301: Build fail on MacOS: feature(asm) may not be used on the stable release channel

MMI, 2021-12-06:
This is probably a local config problem but...

```
➜  hubris git:(master) rustup show
Default host: x86_64-apple-darwin
rustup home:  /Users/georgn/.rustup

installed toolchains
--------------------

stable-x86_64-apple-darwin
nightly-2019-01-19-x86_64-apple-darwin
nightly-2020-04-07-x86_64-apple-darwin
nightly-2021-09-22-x86_64-apple-darwin
nightly-x86_64-apple-darwin

installed targets for active toolchain
--------------------------------------

thumbv7em-none-eabihf
thumbv8m.main-none-eabihf
x86_64-apple-darwin

active toolchain
----------------

nightly-2021-09-22-x86_64-apple-darwin (overridden by '/Users/georgn/git/hubris/rust-toolchain.toml')
rustc 1.57.0-nightly (ac2d9fc50 2021-09-21)

➜  hubris git:(master) cargo xtask dist app/demo-stm32h7-nucleo/app-h7b3.toml
    Finished dev [optimized + debuginfo] target(s) in 0.64s
     Running `target/debug/xtask dist app/demo-stm32h7-nucleo/app-h7b3.toml`
flash = 8000000..8100000
ram = 20000000..20020000
Used:
flash: 0x24100
ram: 0x8000
building path app/demo-stm32h7-nucleo/../../task/jefe
   Compiling cortex-m v0.7.3
   Compiling userlib v0.1.0 (/Users/georgn/git/hubris/sys/userlib)
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> sys/userlib/src/lib.rs:28:1
   |
28 | #![feature(asm)]
   | ^^^^^^^^^^^^^^^^

error[E0554]: `#![feature]` may not be used on the stable release channel
  --> sys/userlib/src/lib.rs:29:1
   |
29 | #![feature(naked_functions)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `userlib` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> /Users/georgn/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/lib.rs:58:45
   |
58 | #![cfg_attr(feature = "inline-asm", feature(asm))]
   |                                             ^^^

error: build failed
Error: failed to build jefe

Caused by:
    command failed, see output for details
```

Any hints or suggestions welcome.

kc8apf, 2021-12-06:
I recall @ahl running into something similar not too long ago.

MMI, 2021-12-06:
@dancrossnyc that feels like it got further but we have a different failure:

```
➜  hubris git:(master) cargo +nightly xtask dist app/demo-stm32h7-nucleo/app-h7b3.toml
   Compiling libc v0.2.105
   Compiling proc-macro2 v1.0.32
   Compiling unicode-xid v0.2.2
[...elided...]
   Compiling num-derive v0.3.3
   Compiling zerocopy v0.3.0
   Compiling ssmarshal v1.0.0
   Compiling abi v0.1.0 (/Users/georgn/git/hubris/sys/abi)
error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:152:9
    |
152 |         sysnum = const Sysnum::Send as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:302:9
    |
302 |         sysnum = const Sysnum::Recv as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:361:9
    |
361 |         sysnum = const Sysnum::Reply as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:424:9
    |
424 |         sysnum = const Sysnum::SetTimer as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:473:9
    |
473 |         sysnum = const Sysnum::BorrowRead as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:533:9
    |
533 |         sysnum = const Sysnum::BorrowWrite as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:599:9
    |
599 |         sysnum = const Sysnum::BorrowInfo as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:636:9
    |
636 |         sysnum = const Sysnum::IrqControl as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:669:9
    |
669 |         sysnum = const Sysnum::Panic as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:749:9
    |
749 |         sysnum = const Sysnum::GetTimer as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: sym operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:820:9
    |
820 |         main = sym main,
    |         ^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_sym)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:894:9
    |
894 |         sysnum = const Sysnum::RefreshTaskId as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

error[E0658]: const operands for inline assembly are unstable
   --> sys/userlib/src/lib.rs:930:9
    |
930 |         sysnum = const Sysnum::Post as u32,
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: see issue #72016 <https://github.com/rust-lang/rust/issues/72016> for more information
    = help: add `#![feature(asm_const)]` to the crate attributes to enable

For more information about this error, try `rustc --explain E0658`.
error: could not compile `userlib` due to 13 previous errors
Error: failed to build jefe

Caused by:
    command failed, see output for details
➜  hubris git:(master)
```

dancrossnyc, 2021-12-06:
Hmm, that feels like a different issue: `cargo +nightly` seems to be invoking an even newer version of the toolchain than what Hubris is currently written to work with. It may be worth trying, `cargo +nightly-2021-09-22 dist ...` and see if that helps. Sadly, I cannot reproduce locally.

kc8apf, 2021-12-06:
A few notes:
* .cargo/confg in the hubris repo aliases xtask to `run --package xtask --`
which runs the program in HUBRIS_ROOT/build/xtask/src/main.rs
* xtask runs `cargo rustc` internally (see build() in build/xtask/src/
dist.rs)
* Hubris uses a rust-toolchain.toml to specify the channel and targets

This all points to `cargo rustc` invoked by `cargo xtask` not using the
rust-toolchain.toml.  I think @ahl had a situation where there were
multiple Rust installs and `cargo` on the command-line ran a different
cargo from the invocation done by `cargo xtask`.  Also check that you have
a new enough rustup that supports the rust-toolchain.toml that specifies
targets.

MMI, 2021-12-06:
@dancrossnyc your suggestion of naming a specific nightly works (at least, I build to completion)...  next step, connect my and flash my board. Thanks!

dancrossnyc, 2021-12-06:
Sure thing, though you should definitely follow the debugging steps Rick outlined; what you're seeing is not the expected behavior.

ahl, 2021-12-06:
@MMI I hit a similar looking (although possibly unrelated) issue that I found to be related to a very very old rustup install (possibly from homebrew). I think it was from 2015 or 2016. I resolved it by ripping out anything related to rustup and cargo root and branch and then reinstalling.

MMI, 2021-12-07:
Unfortunately, I'm a bit of a rust-curious n00b (to borrow Bryan's metaphor)...  so Rick's debugging "steps" are somewhat opaque to me just now...

Of course, suspecting a rust version problem was precisely why I included the `rustup show` output in my original report.

Doing `rustup default nightly-2021-09-22` allows me to use the unadorned `cargo xtask ...` invocation (nightly does not work, as already noted).

dancrossnyc, 2021-12-07:
> Unfortunately, I'm a bit of a rust-curious n00b (to borrow Bryan's metaphor)... so Rick's debugging "steps" are somewhat opaque to me just now...

Nothing wrong with that! :-)

> Of course, suspecting a rust version problem was precisely why I included the `rustup show` output in my original report.
> 
> Doing `rustup default nightly-2021-09-22` allows me to use the unadorned `cargo xtask ...` invocation (nightly does not work, as already noted).

Something I've observed in the past is that `rustup override` in a particular directory will override the toolchain file. A couple of debugging steps that you may give a whirl:

* Run `rustup self update` then `rustup update` to make sure you've got the latest toolchain(s).
* Clear overrides (start with `rustup override list` and try removing a few via `rustup override unset`).

Personally, I stick with `stable` as my default Rust toolchain, and then use rust-toolchain.toml or select `rustup override set`. Note that the latter mechanism came before the former; in general, I eschew overrides these days if I can.

cbiffle, 2021-12-10:
I'd strongly discourage using `rustup default` or `rustup override` because they're hard to notice and easy to forget about. This should be working without additional command line flags, and does on Macs here -- so something's funky.

It sounds like your toolchain is ignoring our `rust-toolchain.toml` file, despite `rustup show` printing the right toolchain. Make sure that the `cargo` you're getting in your path is the one managed by `rustup` -- probably `which cargo` should print `/Users/yourname/.cargo/bin/cargo`. If you get something else, like `/usr/local/bin`, you've got conflicting installs.

`cargo version` executed in the Hubris repo (_without_ any hiding `default`/`override`) should print

```
$ cargo --version
cargo 1.57.0-nightly (9a28ac83c 2021-09-18)
```

MMI, 2021-12-30:
**Thanks for everybody's help.** With some free time over the holidays I revisited...

Turns out that `which cargo` returned the correct thing but there were a handful of links to rust things in my `/usr/local/bin` (presumably from brew) all pointing at 1.56...  So I removed all that and restored my default to stable.

So now the only path to rust things is via `$HOME/.cargo/bin` and alas, the build fails thus:

```
➜  hubris git:(master) ✗ cargo xtask dist app/demo-stm32h7-nucleo/app-h7b3.toml
    Finished dev [optimized + debuginfo] target(s) in 0.20s
     Running `target/debug/xtask dist app/demo-stm32h7-nucleo/app-h7b3.toml`
flash = 8000000..8100000
ram = 20000000..20020000
Used:
flash: 0x24100
ram: 0x8000
building path app/demo-stm32h7-nucleo/../../task/jefe
   Compiling cortex-m v0.7.3
   Compiling userlib v0.1.0 (/Users/georgn/git/hubris/sys/userlib)
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> sys/userlib/src/lib.rs:28:1
   |
28 | #![feature(asm)]
   | ^^^^^^^^^^^^^^^^

error[E0554]: `#![feature]` may not be used on the stable release channel
  --> sys/userlib/src/lib.rs:29:1
   |
29 | #![feature(naked_functions)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0554`.
error: could not compile `userlib` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> /Users/georgn/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/lib.rs:58:45
   |
58 | #![cfg_attr(feature = "inline-asm", feature(asm))]
   |                                             ^^^

error: build failed
Error: failed to build jefe

Caused by:
    command failed, see output for details
➜  hubris git:(master) ✗
```

If I try to explicitly set my default:

```
➜  hubris git:(master) ✗ rustup default nightly
info: using existing install for 'nightly-x86_64-apple-darwin'
info: default toolchain set to 'nightly-x86_64-apple-darwin'

  nightly-x86_64-apple-darwin unchanged - rustc 1.59.0-nightly (78fd0f633 2021-12-29)

info: note that the toolchain 'nightly-2021-09-22-x86_64-apple-darwin' is currently in use (overridden by '/Users/georgn/git/hubris/rust-toolchain.toml')
```

It certainly looks like your `rust-toolchain.toml` is being noticed...

However, this setting breaks the build in other ways (I'll spare you the output but the errors come from `/Users/georgn/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/../asm/inline.rs`.  

That suggests that there is some part(s) of your build that is _not_ noticing the `rust-toolchain.toml`.

For me, it seems that the only path through the labyrinth appears to be `rustup default nightly-2021-09-22-x86_64-apple-darwin` (which does successfully build everything).

cbiffle, 2022-02-01:
We've seen this behavior on Macs a few times and, not having a Mac, I am not sure how it happens. Closing this for now.

MMI, 2022-02-01:
No argument here... it's in the google record now for the next victim 😄 
Thanks again.

closed 2022-02-03 by steveklabnik