oxidecomputer/hubris #304: Minor (e.g. typographical) Errors in Hubris Reference

Kylebrown9, 2021-12-07:
In reading the [Hubris Reference](https://hubris.oxide.computer/reference), I noticed a few very small errors.
This issue may appear long, but that's just because I've tried to be extremely thorough.
I expect these will be very quick to fix.

### Typesetting of `!=` ([Section 4.3](https://hubris.oxide.computer/reference/#_interrupts_from_a_tasks_perspective))

In the code snippet `MY_INTERRUPTS!= 0`, the `!` is typeset as being part of the identifier `MY_INTERRUPTS` instead of as part of the operator `!=`. This is likely an issue with the syntax highlighting tool used, but might be solved by adding a space before the `!`.

Relevant File: [interrupts.adoc](https://github.com/oxidecomputer/hubris/blob/master/doc/interrupts.adoc)

### Typesetting of `<=` ([Section 5.2](https://hubris.oxide.computer/reference/#_programmers_model))

In the following sentence, the less-than-or-equals operator is rendered as a left arrow ligature.
> If not (it is ⇐ the current time), the kernel will

I am not sure how this should be fixed, it appears to be caused by your document generator.
If you can turn off automatic ligature-ing of code snippets, that might be a good idea.

Relevant File: [timers.adoc](https://github.com/oxidecomputer/hubris/blob/master/doc/timers.adoc)

### "interrupt mask" vs "notification mask" ([Section 7](https://hubris.oxide.computer/reference/#_hubris_syscalls))

In the following snippet, the term "interrupt mask" is used.
To my understanding, this is inconsistent with the rest of the document which uses the language "notification mask" .
> In general, the following sorts of things are syscalls (exhaustive):
> ...
> * Control of the current task’s interrupt mask

Relevant File: [syscalls.adoc](https://github.com/oxidecomputer/hubris/blob/master/doc/syscalls.adoc)

cbiffle, 2021-12-13:
Thanks for the close reading of the docs!

- I'll fix the error in section 4.3.
- The unwanted ligature in section 5.2 has been fixed already, but our asciidocs appear not to be auto-publishing; I'll kick off the job, fix should be up shortly.
- The terminology in section 7 is correct, the interrupt mask is controlled by the `IRQ_CONTROL` syscall and is related to, but different from, the notification mask given to `RECV`. Perhaps we could come up with a better term than "interrupt mask" to make the difference more clear.

closed 2021-12-13 by cbiffle

Kylebrown9, 2021-12-14:
@cbiffle the term "interrupt mask" is not used anywhere else throughout the document, which is part of why I was confused. Should a description-of/reference-to it be added somewhere?

commit c43cfcab (2021-12-13): doc/interrupts: fix code example typo

Fixes #304.
PR #328 (2021-12-13): doc/interrupts: fix code example typo
Fixes #304.
commit 4a1b66e0 (2021-12-13): doc/interrupts: fix code example typo

Fixes #304.