oxidecomputer/hubris #307: Link in README to gemini-bringup repository is broken

jsanders, 2021-12-08:
The link in the ["Gemini bringup board"](https://github.com/oxidecomputer/hubris#gemini-bringup-board) section of the README (which points to https://github.com/oxidecomputer/gemini-bringup#getting-started) is broken. 

I suspect this is a repository that is not open to anyone, perhaps on purpose, but I thought it was worth mentioning, in case this is an oversight.

stappersg, 2021-12-08:
On Tue, Dec 07, 2021 at 04:34:24PM -0800, James Sanders wrote:
> I suspect this is a repository that is not open to anyone, perhaps
> on purpose, but I thought it was worth mentioning, in case this is
> an oversight.

It is indeed something that is seen only from "the outside".

I confirm the 404 and say:  Thanks for reporting

cbiffle, 2021-12-10:
Thanks for reporting that! That repo contains the schematics for one of our early test boards, and isn't (currently) public. (It's also not very interesting, speaking as the board designer.)

closed 2021-12-10 by cbiffle