oxidecomputer/hubris #353: I2C target should better deal with ARLO

bcantrill, 2021-12-31:
Exactly once, a Gimlet failed to boot ([dump](https://drive.google.com/file/d/1ugr1X2NOU3_uTuHVtwppEci19ISlwmsV/view?usp=sharing)); we are spinning in `spd`:

```
system time = 2823724
ID TASK                 GEN PRI STATE    
 0 jefe                   0   0 recv, notif: bit0 bit1(T+76)
 1 rcc_driver             0   1 recv
 2 gpio_driver            0   2 recv
 3 spi4_driver            0   2 recv
 4 spi2_driver            0   2 recv
 5 i2c_driver             0   2 recv
 6 spd                    0   2 RUNNING
 7 thermal                0   3 ready
 8 hiffy                  0   3 ready
 9 gimlet_seq             0   3 recv
10 hf                     0   3 recv
11 idle                   0   5 ready
```

The `spd` ring buffer indicates what's happening:

```
humility: ring buffer drv_stm32h7_i2c::__RINGBUF in spd:
 NDX LINE      GEN    COUNT PAYLOAD
  13  849    64115        1 TxISR(0x6c0201)
  14  902    64115        1 WaitTx
  15  849    64115        1 TxISR(0x6c0201)
  16  902    64115        1 WaitTx
  17  849    64115        1 TxISR(0x6c0201)
  18  902    64115        1 WaitTx
  19  849    64115        1 TxISR(0x6c0201)
  20  902    64115        1 WaitTx
  21  849    64115        1 TxISR(0x6c0201)
  22  902    64115        1 WaitTx
  23  849    64115        1 TxISR(0x6c0201)
  24  902    64115        1 WaitTx
  25  849    64115        1 TxISR(0x6c0201)
  26  902    64115        1 WaitTx
  27  849    64115        1 TxISR(0x6c0201)
  28  902    64115        1 WaitTx
  29  849    64115        1 TxISR(0x6c0201)
  30  902    64115        1 WaitTx
  31  849    64115        1 TxISR(0x6c0201)
  32  902    64115        1 WaitTx
  33  849    64115        1 TxISR(0x6c0201)
  34  902    64115        1 WaitTx
  35  849    64115        1 TxISR(0x6c0201)
  36  902    64115        1 WaitTx
  37  849    64115        1 TxISR(0x6c0201)
  38  902    64115        1 WaitTx
  39  849    64115        1 TxISR(0x6c0201)
  40  902    64115        1 WaitTx
  41  849    64115        1 TxISR(0x6c0201)
  42  902    64115        1 WaitTx
  43  849    64115        1 TxISR(0x6c0201)
  44  902    64115        1 WaitTx
  45  849    64115        1 TxISR(0x6c0201)
  46  902    64115        1 WaitTx
  47  849    64115        1 TxISR(0x6c0201)
   0  902    64116        1 WaitTx
   1  849    64116        1 TxISR(0x6c0201)
   2  902    64116        1 WaitTx
   3  849    64116        1 TxISR(0x6c0201)
   4  902    64116        1 WaitTx
   5  849    64116        1 TxISR(0x6c0201)
   6  902    64116        1 WaitTx
   7  849    64116        1 TxISR(0x6c0201)
   8  902    64116        1 WaitTx
   9  849    64116        1 TxISR(0x6c0201)
  10  902    64116        1 WaitTx
  11  849    64116        1 TxISR(0x6c0201)
  12  902    64116        1 WaitTx
```

This denotes ARLO -- arbitrarily lost -- which we don't currently handle.  It's unclear what to do in this case, but the current behavior (spinning taking the interrupt) is definitely wrong.  It seems that it might be best to explicitly panic in this case, and hope that the subsequent `spd` restart and retries from the SP3 would result in a correct read.

cbiffle, 2022-01-05:
Panicking the driver on ARLO is probably not right, because ARLO can happen in a correct system due to e.g. SP3 timing or even electrical noise on the bus causing the I2C block to decide that there's another entity driving it. We should probably stop, wait for bus free, and retry.

lzrd, 2022-01-27:
Wait, 1 retry. SMBUS reset, retry, reset target device and muxes, retry, reboot if all that fails. If this persists, RMA.
If this becomes a long term persistent problem, we will want to trigger a logic analyzer as soon as a problem is noticed. Do we have an SP GPIO pin reserved for triggering? We should have a pin for that.

cbiffle, 2023-04-27:
@bcantrill has done a bunch of work in this area recently -- can we close this?

bcantrill, 2023-04-28:
Yes, in particular I believe that this was fixed as part of #535. Closing.

closed 2023-04-28 by bcantrill