oxidecomputer/hubris #354: Should SPI `read()` send dummy data?

jgallagher, 2022-01-02:
I've been working with a very finicky SPI device attached to an stm32f4, and copy-paste-modified my way from `drv/stm32h7-spi-server` to an SPI server driver for the stm32f4. I couldn't get `read()` to work though, and I based on the comments in the changes in #351, I'm wondering if there's a bug in `stm32h7-spi-server`'s read implementation. It passes `None` as the tx buffer: https://github.com/oxidecomputer/hubris/blob/8e0b13b86564fc7316428943dfe5fde88bb60ef4/drv/stm32h7-spi-server/src/main.rs#L152

which means `ready_writey` will never enter the block that transmits data (dummy or otherwise): https://github.com/oxidecomputer/hubris/blob/8e0b13b86564fc7316428943dfe5fde88bb60ef4/drv/stm32h7-spi-server/src/main.rs#L366-L395

I tested fixing this with a quick hack by adding an "else send dummy data" branch to that block, and it makes `read()` work for me. I doubt that's the best fix, but wanted to open this issue for awareness (or correction if I'm misreading/misunderstanding).

cbiffle, 2022-01-05:
Interesting -- the intent with `read` is that it shifts out dummy data. I'm not sure it's currently being exercised on our side, though. The STM32H7 SPI driver state machine has gotten a little..._hairy_ over the past year or so and I could totally believe that there's a bug here.

cbiffle, 2022-01-22:
@jgallagher you're spot on here, btw. I have a fix prepared (which also greatly improves SPI performance) but it's currently traffic jammed behind a few other of my pending PRs.

cbiffle, 2022-02-02:
This should be fixed on the stm32h7.

closed 2022-02-02 by cbiffle