oxidecomputer/hubris #394: how to make rust-analyzer work with build scripts?

myrfy001, 2022-01-24:
Rust Analyzer will be blocked by some crate's build script which leads to a bad develop experience.

I'm using VSCode as my IDE, and the error hint won;t work because some crate block rust analyzer from working.

for example:

* `sys\userlib\build.rs` will panic
* `sys\num-tasks\build.rs` will panic

is there any tips on how to config RA to walkaround and provide a better dev experience?

thanks~

jeffallen, 2022-01-24:
There is stuff in various Cargo.toml files that mention this problem, for example:
https://github.com/oxidecomputer/hubris/blob/f8851cf677c4710ada599ea176c5e055b6ff6abe/task/template/Cargo.toml#L9

cbiffle, 2022-01-24:
If you're getting panics in those build scripts, you haven't set the target triple correctly. I don't use VSCode, but, change its target setting to something like `thumbv7em-none-eabihf`.

myrfy001, 2022-01-25:
@jeffallen  It seems that there has the `test = false` in `cargo.toml` at `sys/userlib/Cargo.toml`, but RA still tell me `build.rs` panic
https://github.com/oxidecomputer/hubris/blob/06d678143c0c8a3e5799b4e25c33b62bce803f25/sys/userlib/Cargo.toml#L28

@cbiffle when setting target tuple for RA, RA will treat all the crates in the workspace using this target, it seems that there is no easy way to make RA use different target for different crate.

timblakely, 2022-03-19:
I think this _might_ be able to be worked around via the `"rust-analyzer.server.extraEnv": { ... }` VSCode setting. I was hitting the same error @myrfy001 was seeing, and it looks like it was because the environment variable `HUBRIS_TASKS` wasn't being set. Testing out that theory I set `"rust-analyzer.server.extraEnv": { "HUBRIS_TASKS": "foo,bar" }` and it at least passed that step but errored out somewhere in `g0`'s PAC due to missing features, which in turn look like they're _also_ specified in `fn build`. 

Digging into it a bit further it looks like quite a bit of the configuration is set in `fn build` which is called in turn from the various `xtask` invocations. It wouldn't be too far-fetched to write a quick rust app that parses the `app.toml` file and spits out the appropriate `rust-analyzer.*` settings configuration. Mildly annoying is the fact that any change to those settings - or at the very least `rust-analyzer.server.extraEnv` - means having to restart the server and thus the entire window, which makes iterating on the `app.toml` somewhat frustrating. That said, I'm not familiar enough with rust-analyzer to know if there's a way to programmatically observe and update the config.

logerup, 2024-01-25:
I know this isn't a priority, but is there any update or current timeline on this issue or a general working rust-analyzer integration for VS Code?

P2Tree, 2024-12-24:
I encounter a similar issue where my repo has a `build.rs` file located at the root directory of the project, which causes the rust-analyzer to panic when using vscode with rust-analyzer. Delete the file resolves the issue. The problem persists even if clean contents of `build.rs` to empty.
Is it possible that rust-analyzer has a bug when used with vscode in this case?
Moreover, it's worth noting that I have not encountered any issues when using vim or emacs with rust-analyzer.