oxidecomputer/hubris #413: kipc for getting panic messages

cbiffle, 2022-01-31:
Currently, Jefe does not log other tasks' panic messages. We can extract them by Humility, but, that's not a useful solution in production. We will likely eventually want Jefe to be able to extract them and send them off to a logging service. For this, we will need a new kernel API operation alongside the existing `read_task_status`.

If we had that operation, we could also use it in the test suite to test panic message truncation behavior, which is currently not tested.

jeffallen, 2022-02-01:
Why would this not just be part of what `read_task_status` returns? In fact, there's already `FaultInfo::Panic` in the ABI.

cbiffle, 2023-04-27:
Finally getting back around to this --

The reason the panic message isn't in `read_task_status` is that it's variable-length and may be missing, which is problematic for the serde format we currently use for serializing kipc responses. That limitation is not _inherent_ however, and we could maybe fix that. That would require diverging `TaskState` as returned from the kipc from the actual `TaskState` used in the kernel, however, since the kernel's `TaskState` _does not_ include the panic message. Instead, the panic message lives in task RAM.

I have a feeling this is about to come up on the priority radar again soon.

closed 2023-04-27 by cbiffle