oxidecomputer/hubris #439: "Insufficient flash" error message doesn't always seem to work

steveklabnik, 2022-02-24:


steveklabnik, 2022-02-24:
As an example, after fixing the `app.toml` to have the right amount for `task-net`, here's the next error I got:

```
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "C:\\Users\\steve\\Documents\\Oxide\\hubris\\target\\thumbv7em-none-eabihf\\release\\deps\\task_udpecho-25fcf3909630cdca.task_udpecho.740dc942-cgu.0.rcgu.o" "--as-needed" "-L" "C:\\Users\\steve\\Documents\\Oxide\\hubris\\target\\thumbv7em-none-eabihf\\release\\deps" "-L" "C:\\Users\\steve\\Documents\\Oxide\\hubris\\target\\release\\deps" "-L" "target" "-L" "C:\\Users\\steve\\.rustup\\toolchains\\nightly-2022-02-22-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "--start-group" "--end-group" "-Bstatic" "C:\\Users\\steve\\.rustup\\toolchains\\nightly-2022-02-22-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib\\libcompiler_builtins-e8b9a93ab984480c.rlib" "-Bdynamic" "--eh-frame-hdr" "-znoexecstack" "-L" "C:\\Users\\steve\\.rustup\\toolchains\\nightly-2022-02-22-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "-o" "C:\\Users\\steve\\Documents\\Oxide\\hubris\\target\\thumbv7em-none-eabihf\\release\\deps\\task_udpecho-25fcf3909630cdca" "--gc-sections" "-Tlink.x" "-z" "common-page-size=0x20" "-z" "max-page-size=0x20"
  = note: rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 57 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 76 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 92 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 118 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 136 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 152 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 168 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 184 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 200 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 216 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 232 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 275 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 292 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 292 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 336 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 364 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 380 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 388 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 412 bytes
          rust-lld: error: section '.rodata' will not fit in region 'FLASH': overflowed by 412 bytes
          rust-lld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
```

cbiffle, 2023-04-27:
Can definitely confirm that this still doesn't always work.