oxidecomputer/hubris #450: Error: environment variable not found

Shengliang, 2022-03-14:
Hi,

I got the following error?  Any hints?  Any document to add a new board (or CPU) to hubris? 

Thanks,
Sheng-Liang 
PS:

Setup: Surface-laptop 4 load with ubuntu 21.10: 
Linux Surface-Laptop-4 5.13.0-35-generic #40-Ubuntu SMP Mon Mar 7 08:03:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

cargo xtask dist app/demo-stm32h7-nucleo/app-h743.toml
    Finished dev [optimized + debuginfo] target(s) in 0.16s
     Running `target/debug/xtask dist app/demo-stm32h7-nucleo/app-h743.toml`
flash = 0x08000000..0x08100000
ram   = 0x20000000..0x20020000
sram1 = 0x30000000..0x30020000
Used:
  flash: 0x32000 (19%)
  ram:   0x10000 (50%)
  sram1: 0x8000 (25%)
Error: environment variable not found

kc8apf, 2022-03-14:
Hmm.  That error message doesn't exist in Hubris' xtask so it must be coming from a tool invocation being done by xtask.  Hubris' xtask does use environment variables to pass information to various cargo invocations.  What does `cargo xtask dist -v app/demo-stm32h7-nucleo/app-h743.toml` tell you?

kc8apf, 2022-03-14:
So it's dying sometime after build/xtask/src/dist.rs:135. There is a call to std::env::var() soon after but it's for a CARGO_HOME which should be set by cargo when xtask is invoked.  What cargo version are you using?

Shengliang, 2022-03-14:
ssl@ssl-Surface-Laptop-4:~/hubris$ cargo --version
cargo 1.57.0
ssl@ssl-Surface-Laptop-4:~/hubris$ which cargo
/usr/bin/cargo
ssl@ssl-Surface-Laptop-4:~/hubris$ rustc --version
rustc 1.57.0

Shengliang, 2022-03-15:
FYI:  Resolved with 1.59.0 version.

Thanks!

ssl@ssl-Surface-Laptop-4:~$ which cargo
/home/ssl/.cargo/bin/cargo
ssl@ssl-Surface-Laptop-4:~$ cargo --version
cargo 1.59.0 (49d8809dc 2022-02-10)
ssl@ssl-Surface-Laptop-4:~$ rustc --version
rustc 1.59.0 (9d1b2106e 2022-02-23)
ssl@ssl-Surface-Laptop-4:~$ which rustc
/home/ssl/.cargo/bin/rustc

closed 2022-03-15 by Shengliang

Shengliang, 2022-03-15:
FYI. 

ssl@ssl-Surface-Laptop-4:~/hubris$ cargo xtask test test/tests-stm32h7/app-h743.toml
    Finished dev [optimized + debuginfo] target(s) in 0.15s
     Running `target/debug/xtask test test/tests-stm32h7/app-h743.toml`
flash = 0x08000000..0x08100000
ram   = 0x20000000..0x20020000
Used:
  flash: 0x20100 (12%)
  ram:   0x4100 (12%)
building path test/tests-stm32h7/../test-runner
    Finished release [optimized + debuginfo] target(s) in 0.14s
target/thumbv7em-none-eabihf/release/test-runner -> target/tests-stm32h743/dist/runner
building path test/tests-stm32h7/../test-suite
    Finished release [optimized + debuginfo] target(s) in 0.14s
target/thumbv7em-none-eabihf/release/test-suite -> target/tests-stm32h743/dist/suite
building path test/tests-stm32h7/../test-assist
    Finished release [optimized + debuginfo] target(s) in 0.14s
target/thumbv7em-none-eabihf/release/test-assist -> target/tests-stm32h743/dist/assist
building path test/tests-stm32h7/../../task/idle
    Finished release [optimized + debuginfo] target(s) in 0.14s
target/thumbv7em-none-eabihf/release/task-idle -> target/tests-stm32h743/dist/idle
building path test/tests-stm32h7/../../app/demo-stm32h7-nucleo
    Finished release [optimized + debuginfo] target(s) in 0.14s
target/thumbv7em-none-eabihf/release/demo-stm32h7-nucleo -> target/tests-stm32h743/dist/kernel
Open On-Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1800 kHz
Info : STLINK V3J4M2 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.294618
Info : stm32h7x.cpu0: hardware has 8 breakpoints, 4 watchpoints
Info : starting gdb server for stm32h7x.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000298 msp: 0x20000400
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
Info : Unable to match requested speed 4000 kHz, using 3300 kHz
** Programming Started **
Info : Device: STM32H74x/75x
Info : flash size probed value 2048
Info : STM32H7 flash has dual banks
Info : Bank (0) size is 1024 kb, base address is 0x08000000
Info : Padding image section 0 at 0x08005014 with 12268 bytes
Info : Padding image section 1 at 0x0800b090 with 3952 bytes
Info : Padding image section 2 at 0x0800e2f0 with 7440 bytes
Info : Padding image section 3 at 0x08018608 with 31224 bytes
Info : Padding image section 4 at 0x08020058 with 8 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x08020060 .. 0x0803ffff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
humility: attached via ST-Link V3
humility: ITM synchronization packet found at offset 6
humility: expecting 42 cases
humility: running test_send ... ok
humility: running test_recv_reply ... ok
humility: running test_recv_reply_fault ... ok
humility: running test_floating_point_lowregs ... ok
humility: running test_floating_point_highregs ... ok
humility: running test_floating_point_fault ... ok
humility: running test_fault_badmem ... ok
humility: running test_fault_stackoverflow ... ok
humility: running test_fault_execdata ... ok
humility: running test_fault_illop ... ok
humility: running test_fault_nullexec ... ok
humility: running test_fault_textoob ... ok
humility: running test_fault_stackoob ... ok
humility: running test_fault_buserror ... ok
humility: running test_fault_illinst ... ok
humility: running test_fault_divzero ... ok
humility: running test_fault_maxstatus ... ok
humility: running test_fault_badstatus ... ok
humility: running test_fault_maxrestart ... ok
humility: running test_fault_badrestart ... ok
humility: running test_fault_maxinjection ... ok
humility: running test_fault_badinjection ... ok
humility: running test_fault_superinjection ... ok
humility: running test_fault_selfinjection ... ok
humility: running test_panic ... ok
humility: running test_restart ... ok
humility: running test_restart_taskgen ... ok
humility: running test_borrow_info ... ok
humility: running test_borrow_read ... ok
humility: running test_borrow_write ... ok
humility: running test_borrow_without_peer_waiting ... ok
humility: running test_supervisor_fault_notification ... ok
humility: running test_timer_advance ... ok
humility: running test_timer_notify ... ok
humility: running test_timer_notify_past ... ok
humility: running test_task_status ... ok
humility: running test_task_fault_injection ... ok
humility: running test_refresh_task_id_basic ... ok
humility: running test_refresh_task_id_off_by_one ... ok
humility: running test_refresh_task_id_off_by_many ... ok
humility: running test_lpc55_flash_write ... ok
humility: running test_post ... ok
humility: tests completed: pass
ssl@ssl-Surface-Laptop-4:~/hubris$

therishidesai, 2022-09-25:
I'm getting this error as well but it doesn't even seem to be getting as far as your thing got.

Using rutsc 1.64.0 and cargo 1.64.0
```
[nix-shell:~/hubris]$ cargo xtask dist app/demo-stm32f4-discovery/app-f3.toml
    Finished dev [optimized + debuginfo] target(s) in 0.24s
     Running `target/debug/xtask dist app/demo-stm32f4-discovery/app-f3.toml`
Error: environment variable not found
```