oxidecomputer/hubris #454: Reinstate support for STM32H7B3I-DK?

rumpuslabs, 2022-03-16:
In ee355435f235bbb98c7ac252807d7154140c4c52 @cbiffle started removing support for the H7B3 with @bcantrill's approval, unfortunately I didn't notice this and, since it was the only STM32 board I could find in stock anywhere reputable, I bought one.

Octopart searches:
* [STM32F303](https://octopart.com/search?q=STM32F3discovery)
* [STM32F407](https://octopart.com/search?q=STM32F407+discovery)
* [STM32H743](https://octopart.com/search?q=STM32H743+nucleo)
* [STM32H753](https://octopart.com/search?q=STM32H753+nucleo)
* [STM32H7B3](https://octopart.com/search?q=STM32H7B3I-DK)

Would there be any objections or impediments to me reimplementing support for this board?

cbiffle, 2022-03-16:
Hi! I started removing the 7B3 support after talking about doing so for some months; it was very disruptive for a variety of reasons, such as the SVDs for the 7B3/753 differing in names that the upstream PACs picked up in their APIs, slight difference in hardware setup and errata, etc. We're unlikely to use the 7B3 in a product; while in many ways it's a substantially regularized and improved 753, it also lacks Ethernet, which makes it a non-starter for us. This meant we had a large maintenance/support burden with no clear justification.

(ST, if you're listening, how about a 7B3 descendant with dual Ethernet MACs? Love, Cliff.)

Depending on what you're looking to do with Hubris, there are other part lines you might consider, though they are effectively "second-tier" supported -- we don't have formal support tiers at the moment, but I mean that they're not currently being used in a product and don't have monitored test coverage. The F411 keeps showing up as available, which should be a small delta relative to the F407 port. As of fairly recently we've got STM32G0 support. STM32L0 support is also mostly done but I appear to have forgotten to merge the branch. The G0 and L0 have been pretty easy to come by lately, though often in annoying packages. (...which is probably why they're in stock.)

I sympathize with your being affected by the parts shortage -- as you might imagine, this is an interesting time to be building a computer.

rumpuslabs, 2022-03-17:
Thanks Cliff. I've submitted a pull request to remove the H7B3 from the docs (and optionally the code as well) to prevent others making the same mistake I did.

I am looking at availability of evaluation boards, not bare parts, as my goal at this stage is just to hack on Hubris. [F411 discovery](https://octopart.com/search?q=STM32F411%20discovery) boards are not available, but the [F412](https://octopart.com/search?q=STM32F412G-DISCO) is. Also [a variety of H7s are](https://octopart.com/search?q=STM32H7+discovery) including H735G, H745I, H747I, and H750B.

Unfortunately none of those are an exact match for the parts you're using. The [H743I-EVAL and H753I-EVAL2](https://octopart.com/search?q=STM32H7+eval) are available but these are significantly more expensive.

The F412 looks like a superset of the F411's features and even closer perhaps to the F407, so I will try one of those unless you think getting a H743 or H753 will make things significantly easier (for me and those reviewing my pull requests).

closed 2023-04-27 by cbiffle