oxidecomputer/hubris #467: lpc55 UART broken

flihp, 2022-03-22:
Ping stopped pinging on the LPC55 recently. I'm pretty sure I've tracked this back to 03a6a4f1. The UART can be restored by setting `secure-separation` to `false` and disabling the `tz_support` feature in the bootloader config.

labbott, 2022-03-22:
The underlying issue is that we need to be setting `NVIC_ITNS` to route interrupts to the non-secure world. Fixing that has also exposed that we need to be more deliberate about our faulting tests since NULL (aka address 0x0) is secure now and we do not have a good way to let hubris gracefully handle a secure fault.

labbott, 2022-03-22:
https://github.com/oxidecomputer/hubris/pull/468 should fix this (as well as a few other issues). I should go file a separate issue for figuring out what exactly we should do about SecureFaults

flihp, 2022-03-23:
works for me. thanks!

closed 2022-03-23 by flihp