oxidecomputer/hubris #487: OpenOCD fallback for flashing

cbiffle, 2022-03-31:
(This may already exist, I'm filing this so it doesn't get lost in the chat logs.)

We've recently switched `xtask flash` to use probe-rs. That's nice for folks who don't have OpenOCD installed, but limits us to supporting the set of chips that probe-rs has correctly implemented flashing on.

When I'm porting to a new (and particularly recently released) SoC, I often wind up writing the flashing sequence by hand and supplying it to OpenOCD as a TCL file. Probe-rs has no equivalent user scripting interface or way of installing SoCs as plugins without modifying upstream, so doing the equivalent change would require forking, patching it, and rebuilding humility.

I'd prefer to be able to flash with OpenOCD on request, if possible.

labbott, 2022-03-31:
I'm not sure I support this for `xtask flash`. Part of the reason for wanting to migrate away from OpenOCD is that updates are historically slow to say the least. There is no support for the LPC55 on OpenOCD and no support for the H7 in pyOCD. Both are supported in probe-rs. I get that OpenOCD is nice for porting but that's also not our highest priority at the moment. I'd also argue that if there is a chip we seriously want to use we _should_ be adding support for it in probe-rs. I'd rather keep debugger flashing support as a separate command and have `xtask flash` covered by what is available in probe-rs.

cbiffle, 2023-04-27:
We did wind up maintaining openocd as a backup option, closing this.

closed 2023-04-27 by cbiffle