oxidecomputer/hubris #513: Are ethernet rings + DMA cache-safe?

mkeeter, 2022-04-25:
In our netstack, we allocate [packet and descriptor buffers in SRAM1](https://github.com/oxidecomputer/hubris/blob/master/task/net/src/buf.rs#L70-L99), then read / write to them through DMA.

In addition, we enable the D-cache in `stm32h7-startup`.

I'm suspicious that we may not be safe with respect to this cache.

I think SRAM1 access happens through the AXIM → D1-to-D2 AHB bus → AHB Bus Matrix → SRAM1 (see Table 3 in [RM0433](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf), which tells you that the AHBP link doesn't have access to SRAM1):

[image]

This means that these accesses are cached by the L1 cache (`D$` in the Cortex-M7 block).

We're not doing anything kind of cache maintenance between DMA and Cortex-M7 access.  I haven't seen any issues _yet_, but suspect that for correctness's sake, we should either
- Exclude that block of RAM from the D-cache
- Add appropriate barriers to the Tx/RxRing code

cbiffle, 2022-04-25:
@labbott that was certainly the intention -- that the Ethernet memory be mapped uncached -- but having more eyes on it is always a good idea.

mkeeter, 2022-04-25:
Cool, I didn't find that part of the code in my investigation.  Looking at Table 91 in PM0253, those look like the correct values for `TEX` and `SCB`!

closed 2022-04-25 by mkeeter