oxidecomputer/hubris #544: STM32H7 I2C: target code sometimes fails to handle all interrupt flags, causing busyloop

cbiffle, 2022-05-10:
When we raised `spd`'s priority on gimlet we noticed that other things were getting starved, which seems darn suspicious given that spd should be spending like 98.7% of its life waiting for interrupts.

Unless it's not waiting.

Inspection of the ringbufs showed a lot of this:

```
  11  846    30017        1 TxISR(0x6c0005)
```

Bit 2 is set, which is RXNE, which is bad. The Tx loop doesn't _handle_ RXNE. So, it is immediately popping out of wfi. The transmit still completes, but only after having left a bloody trail of preemption behind it.

I believe the problem is in the rx loop, specifically around [here](https://github.com/oxidecomputer/hubris/blob/27cb5cd4d7bf118037c8733095f639bb580979b9/drv/stm32h7-i2c/src/lib.rs#L811). If we enter that loop with RXNE set _and_ either STOPF or NACKF, we will handle the STOPF or NACKF and bail before handling the RXNE.

Generally, when responding to an interrupt, we need to make sure to clear _all_ relevant flags before moving on, which in this case means that the loop needs to potentially deal with _all three_ before leaving the loop, rather than early exiting.

(@bcantrill fyi)

cbiffle, 2022-05-11:
This has happened to @rmustacc without humility dashboard running, in response to a PSP scan of the (unproxied) DIMM temperature sensors.

wesolows, 2022-06-09:
I believe I've hit another instance of this on 471633f3d929d2bc3a0daf96b95cb3b55c10f511.  I have:

```
$ humility -d hubris.core.0 ringbuf spd
[2022-06-09T20:09:20Z WARN  humility] dump on command-line overriding archive in environment
humility: attached to dump
humility: ring buffer drv_stm32h7_i2c::__RINGBUF in spd:
 NDX LINE      GEN    COUNT PAYLOAD
  40  847    29350        1 TxISR(0x360005)
  41  900    29350        1 WaitTx
  42  847    29350        1 TxISR(0x360005)
  43  900    29350        1 WaitTx
  44  847    29350        1 TxISR(0x360005)
  45  900    29350        1 WaitTx
  46  847    29350        1 TxISR(0x360005)
  47  900    29350        1 WaitTx
   0  847    29351        1 TxISR(0x360005)
   1  900    29351        1 WaitTx
   2  847    29351        1 TxISR(0x360005)
   3  900    29351        1 WaitTx
   4  847    29351        1 TxISR(0x360005)
   5  900    29351        1 WaitTx
   6  847    29351        1 TxISR(0x360005)
   7  900    29351        1 WaitTx
   8  847    29351        1 TxISR(0x360005)
   9  900    29351        1 WaitTx
  10  847    29351        1 TxISR(0x360005)
  11  900    29351        1 WaitTx
  12  847    29351        1 TxISR(0x360005)
  13  900    29351        1 WaitTx
  14  847    29351        1 TxISR(0x360005)
  15  900    29351        1 WaitTx
  16  847    29351        1 TxISR(0x360005)
  17  900    29351        1 WaitTx
  18  847    29351        1 TxISR(0x360005)
  19  900    29351        1 WaitTx
  20  847    29351        1 TxISR(0x360005)
  21  900    29351        1 WaitTx
  22  847    29351        1 TxISR(0x360005)
  23  900    29351        1 WaitTx
  24  847    29351        1 TxISR(0x360005)
  25  900    29351        1 WaitTx
  26  847    29351        1 TxISR(0x360005)
  27  900    29351        1 WaitTx
  28  847    29351        1 TxISR(0x360005)
  29  900    29351        1 WaitTx
  30  847    29351        1 TxISR(0x360005)
  31  900    29351        1 WaitTx
  32  847    29351        1 TxISR(0x360005)
  33  900    29351        1 WaitTx
  34  847    29351        1 TxISR(0x360005)
  35  900    29351        1 WaitTx
  36  847    29351        1 TxISR(0x360005)
  37  900    29351        1 WaitTx
  38  847    29351        1 TxISR(0x360005)
  39  900    29351        1 WaitTx
humility: ring buffer task_spd::__RINGBUF in spd:
 NDX LINE      GEN    COUNT PAYLOAD
   2  338     2390        1 MuxState(Idle, AwaitingSegment)
   3  316     2390        1 Rx(0x4a, 0x40)
   4  338     2390        1 MuxState(AwaitingSegment, Done)
   5  311     2390        4 Initiate(0x1d, false)
   6  311     2390        1 Initiate(0x4a, true)
   7  316     2390        1 Rx(0x4a, 0x3)
   8  338     2390        1 MuxState(Idle, AwaitingSegment)
   9  316     2390        1 Rx(0x4a, 0x40)
  10  338     2390        1 MuxState(AwaitingSegment, Done)
  11  311     2390        4 Initiate(0x19, false)
  12  311     2390        1 Initiate(0x4a, true)
  13  316     2390        1 Rx(0x4a, 0x3)
  14  338     2390        1 MuxState(Idle, AwaitingSegment)
  15  316     2390        1 Rx(0x4a, 0x40)
   0  338     2391        1 MuxState(AwaitingSegment, Done)
   1  311     2391        4 Initiate(0x1b, false)
```
and
```
$ humility -d hubris.core.0 tasks
[2022-06-09T20:09:49Z WARN  humility] dump on command-line overriding archive in environment
humility: attached to dump
system time = 21999617
ID TASK                 GEN PRI STATE
 0 jefe                   0   0 recv, notif: bit0 bit1(T+83)
 1 net                    0   5 ready
 2 sys                    0   1 recv
 3 spi4_driver            0   3 recv
 4 spi2_driver            0   3 recv
 5 i2c_driver             0   3 recv
 6 spd                    0   2 RUNNING
 7 thermal                0   5 ready
 8 power                  0   6 ready
 9 hiffy                  0   5 ready
10 gimlet_seq             0   4 recv
11 hf                     0   3 recv
12 sensor                 0   4 ready
13 udpecho                0   6 notif: bit0
14 udpbroadcast           0   6 ready
15 validate               0   5 recv
16 idle                   0   7 ready
```
which is still the state it's in.  This was detected by the fact that `humility spi` times out on a command that has previously worked with this software revision and in fact many times on this boot.  Dump is attached.
[hubris.core.0.gz](https://github.com/oxidecomputer/hubris/files/8873589/hubris.core.0.gz)

cbiffle, 2022-06-09:
FWIW the `0x360005` status that we appear to be spinning on decodes as

- I2C device address `0b0011_011`
- Write
- TXE (transmit shift register empty)
- RXNE (receive shift register not empty)
- All other flags clear.

The driver enables interrupt notification on both of those latter bits, but in [the loop where you're spinning](https://github.com/oxidecomputer/hubris/blob/master/drv/stm32h7-i2c/src/lib.rs#L845) it doesn't check or clear either of them.

So, this does appear to be the same bug, thanks for providing more supporting data!

To fix this the I2C driver overall should only set the interrupt enables for events it's prepared to handle (e.g. for this loop, that's addr/nack/txis only). Currently it sets the interrupt enables on driver setup and turns them all on.

There's an open question of why RXNE is set -- sort of implies that the driver failed to read a byte from a previous transaction? @bcantrill is the expert on the state machine here.

commit bc934907 (2022-06-10): stm32h7-i2c: rework target state machine

This rework is focused on handling interrupt flags comprehensively and
at the right times, to try and fix #544 and related issues.
commit 0a32e353 (2022-06-10): stm32h7-i2c: rework target state machine

This rework is focused on handling interrupt flags comprehensively and
at the right times, to try and fix #544 and related issues.
commit 00e9e982 (2022-06-14): stm32h7-i2c: rework target state machine

This rework is focused on handling interrupt flags comprehensively and
at the right times, to try and fix #544 and related issues.
commit b6fa140f (2022-06-20): stm32h7-i2c: rework target state machine

This rework is focused on handling interrupt flags comprehensively and
at the right times, to try and fix #544 and related issues.
closed 2022-06-20 by cbiffle