oxidecomputer/hubris #577: lpc55: disabling TZ support causes jefe to fault

flihp, 2022-05-26:
I've been messing around with the uart in stage0 and, in trying to debug some odd behavior, I disabled the "tz_support" feature. Eventually I realized that just disabling this feature causes the 'jefe' task to fault. This should be reproducible with the following steps:

1. remove the feature from `app/lpc55expresso/app.toml`
2. build & flash the image
3. boot up to hubris
4. list the tasks with humility, the output should look something like:

```bash
    Finished dev [optimized + debuginfo] target(s) in 0.50s
     Running `target/debug/xtask humility app/lpc55xpresso/app.toml -- tasks`
humility: attached via CMSIS-DAP
system time = 0
ID TASK                 GEN PRI STATE    
 0 jefe                   0   0 FAULT: general fault, cfsr=0x40000 (was: ready)
 1 hiffy                  0   5 ready
 2 idle                   0   7 ready
 3 syscon_driver          0   2 RUNNING
 4 gpio_driver            0   3 ready
 5 user_leds              0   4 ready
 6 usart_driver           0   4 ready
 7 i2c_driver             0   4 ready
 8 rng_driver             0   3 ready
 9 spi0_driver            0   4 ready
10 ping                   0   6 ready
11 pong                   0   5 ready
```
LKM if you need any additional data.

cbiffle, 2022-05-27:
FWIW - The ARMv7-M Architecture Reference Manual section B3.2.15 gives the definitions for the CFSR bits. In this case, you've got a usage fault, with the INVPC bit set. (Yeah, I need to improve the fault decoding from humility tasks.) INVPC indicates "An integrity check error has occurred on EXC_RETURN."

So, the fault most likely happens on the very first user instruction, when the kernel attempts to go into userland by this branch [here](https://github.com/oxidecomputer/hubris/blob/master/sys/kern/src/arch/arm_m.rs#L889). The constant it's using to effect the return-to-user is chosen during the build [here](https://github.com/oxidecomputer/hubris/blob/master/sys/kern/build.rs#L32), and appears to be choosing the wrong value if you're getting INVPC. So my inference is that the `HUBRIS_SECURE` setting passed to the kernel build does not match the actual state of the machine.

Likely, your kernel is being built under the assumption that it's running in nonsecure mode, but your disabling TZ in the bootloader means it's actually running in secure mode, and things are getting haywired. You will likely need to change a kernel configuration value to match the bootloader. (And perhaps we should be choosing that configuration value automatically to remove this potential mismatch!)

Past that, I have exceeded the limits of my cached knowledge -- it looks like `HUBRIS_SECURE` is set by a `secure_separation` key in the TOML but I can't find any instances of that in any apps. @labbott will know more.

flihp, 2022-05-27:
I haven't got this deep into the architecture reference, the kernel, or the build yet :sweat_smile:  Good opportunity for me to do some digging. Thanks for the hints / breadcrumbs. Your theory w/r/t coordination between kernel & bootloader makes sense. I'll see about finding the specifics and will report back.

labbott, 2022-05-31:
@cbiffle has the right idea. You also need to turn off `secure-separation` which is on in the xpresso TOML. Changing it two places is awkward and I'm trying to recall why I did it that way. I'd certainly welcome a change to make it less error prone.

cbiffle, 2023-04-27:
We've changed how TZ support works (currently: it does not work) so this particular gotcha is obsolete.

closed 2023-04-27 by cbiffle