oxidecomputer/hubris #590: gimlet: want visual indicator if installed DIMMs don't match a valid memory config

kc8apf, 2022-06-07:
Putting my rev B back together, I installed all the DIMMs in the x0 slots.  The system didn't boot.  Checking Milan's memory population guide, I should have put them in the x1 slots instead.  Since spd task knows which DIMM slots are populated, it'd be really helpful if it could check that the DIMM configuration is a valid config and, if not, blink an LED or something.

cbiffle, 2022-06-07:
I am pleased to report that we _have a fricking LED now_ and could potentially implement this! How complex are the rules?

rmustacc, 2022-06-07:
For the next spin we're attempting to change the actual silk to help here. Obviously that's not on its own enough. @cbiffle the set of legal states isn't too hard to detect in this case.

So an LED to this purpose exactly long term in the product doesn't make sense to me per se because that would be internal to the chassis and you'd have to pull it out of the rack. However, this is a case where I'd expect the control plane to be able to ask the SP to blink the chassis attention LED and give a more detailed message.

kc8apf, 2022-06-07:
SP could just directly blink the chassis attention LED in this case.

cbiffle, 2022-06-07:
Right, I feel like we could (1) detect this without going through the control plane and (2) do something useful to engineers now, with a server on their bench, that may or may not live into the actual PVT deployments. I'm cool with that. So, how would the logic need to go?

rmustacc, 2022-06-07:
The approximate rule for engineering is that any of [A-H]0 are populated without [A-H]1 being populated. For the product we are hoping it'll be more constrained. More formally it's defined by the SP3 memory population guidelines for milan, but I suspect the above is a useful enough approximation.

cbiffle, 2023-04-27:
I'm pretty sure we never actually got around to doing this. Is this still a thing we want, @rmustacc / @kc8apf ?

kc8apf, 2023-04-27:
Yes but definitely not a PVT1 or MVP blocker.