oxidecomputer/hubris #598: `sensors` task improvements

cbiffle, 2022-06-10:
- The sensors task should maintain a timestamp on each reading, so that observers can tell how old the value is. (This is important for control loops and things like integrators.)
- The timestamp must be provided _by the task making the measurement,_ not by the sensors task, to ensure that it's right. Problems that this avoids:
  - Task being preempted between measuring something and getting around to posting it to sensors
  - Due to a simple implementation or a bug, a task repeatedly posting the same measurement such that it looks like it's fresh whether or not it actually is.
- The IPC API should return the timestamp with any measurement so that callers can decide whether to discard the measurement.
- Tasks reading sensors _should not_ be able to replace their last measurement with "no data." If there is no new data, don't post with an updated timestamp. This ensures that we can always observe at least the most recent valid reading, and e.g. limp along even if every other transaction fails.

@kc8apf - anything else?

kc8apf, 2022-06-10:
I'd add a 2nd timestamp to track the last attempt to post a measurement.  That helps distinguish whether a stale measurement is due to the source task not attempting or attempting and failing.

cbiffle, 2022-06-10:
Ooooh that's a good one. So track both "indicated measurement time" and "posting time." Can do.

I should note that my assumption is that all timestamps are in the local processor monotonic timebase, i.e. taken from the Hubris kernel clock. There are various methods for mapping this to walltime if/when we require it, but that simplifies the implementation considerably.

kc8apf, 2022-06-10:
Agree on using SP's local monotonic timebase.  Early boot won't have any notion of what walltime is anyway.

mkeeter, 2025-07-01:
This is mostly done – we now track timestamped data and errors, along with details on our last reading (whether it was data or error):
```rust
struct ServerImpl {
    last_reading: SensorArray<Option<LastReading>>,

    data_value: SensorArray<f32>,
    data_time: SensorArray<u64>,

    min_value: SensorArray<f32>,
    min_time: SensorArray<u64>,

    max_value: SensorArray<f32>,
    max_time: SensorArray<u64>,

    err_value: SensorArray<NoData>,
    err_time: SensorArray<u64>,

    nerrors: SensorArray<u32>,
}
```

We have _not_ implemented separate times for measurement versus posting.  I don't think this is necessary: 100% of tasks use the `post_now` helper function to send data to `sensors`, which posts with the current timestamp.