oxidecomputer/hubris #604: Would like mainline gimlet-reboot-loop

rmustacc, 2022-06-15:
During bring up we started the `gimlet-reboot-loop` branch which has some features for rebooting a gimlet after it signals it came up which is useful for testing DIMM training, running mbist, and other bits. It'd be nice if there was a way to have this as an optional thing we can leverage without maintaining a separate branch and trying to keep it up to date.

cbiffle, 2023-04-27:
@rmustacc is this need still unmet?

jclulow, 2023-04-27:
I think we've mostly solved this in other ways now that we have fleshed out control from outside over both the SWD and Ethernet interfaces.  There is a bootloop tool in lengths of wire that we've used into the tens of thousands of cycles now in rack2.

rmustacc, 2023-04-28:
I think per @jclulow's comment and the tooling we have, this path is no longer needed. Closing this out.

closed 2023-04-28 by rmustacc