oxidecomputer/hubris #657: SPD missing DIMMs

wesolows, 2022-07-20:
On Gimlet-B with rev e3f699d534aa52567182718352af51af5b0de37f, I observed the following after entering A2 but before entering A0:

```
$ humility spd
humility: attached via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   9 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  11 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  13 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  15 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
```
There are in fact 8 DIMMs, so 4 of them are missing.  A dump (after later entering A0) is attached.
[Uploading hubris.core.0.gz…]()

wesolows, 2022-08-06:
A second observation, though we almost certainly haven't enough history in the ringbufs:

```
$ humility spd
humility: attached via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   9 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  11 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  13 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  15 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
wesolows at monolith in ~
$ humility ringbuf spd
humility: attached via ST-Link V3
humility: ring buffer drv_stm32h7_i2c::__RINGBUF in spd:
 NDX LINE      GEN    COUNT PAYLOAD
  36  906     1448        1 WaitRx
  37  827     1448        1 RxISR(0x368001)
  38  906     1448        1 WaitRx
  39  827     1448        1 RxISR(0x368005)
  40  854     1448        1 RxNack(0x1b, 0x0)
  41  827     1448        1 RxISR(0x368001)
  42  906     1448        1 WaitRx
  43  827     1448        1 RxISR(0x368001)
  44  906     1448        1 WaitRx
  45  827     1448        1 RxISR(0x368005)
  46  854     1448        1 RxNack(0x1b, 0xa0)
  47  827     1448        1 RxISR(0x368001)
   0  906     1449        1 WaitRx
   1  827     1449        1 RxISR(0x360021)
   2  866     1449        1 Stop
   3  764     1449        1 AddrISR(0x360001)
   4  787     1449        1 WaitAddr
   5  764     1449        1 AddrISR(0x360001)
   6  787     1449        1 WaitAddr
   7  764     1449        1 AddrISR(0x368009)
   8  779     1449        1 AddrMatch
   9  812     1449        1 AddrNack(0x1b)
  10  827     1449        1 RxISR(0x368001)
  11  906     1449        1 WaitRx
  12  827     1449        1 RxISR(0x368001)
  13  906     1449        1 WaitRx
  14  827     1449        1 RxISR(0x368005)
  15  854     1449        1 RxNack(0x1b, 0x1)
  16  827     1449        1 RxISR(0x368001)
  17  906     1449        1 WaitRx
  18  827     1449        1 RxISR(0x368001)
  19  906     1449        1 WaitRx
  20  827     1449        1 RxISR(0x368005)
  21  854     1449        1 RxNack(0x1b, 0x4)
  22  827     1449        1 RxISR(0x368001)
  23  906     1449        1 WaitRx
  24  827     1449        1 RxISR(0x368001)
  25  906     1449        1 WaitRx
  26  827     1449        1 RxISR(0x368005)
  27  854     1449        1 RxNack(0x1b, 0x8)
  28  827     1449        1 RxISR(0x368001)
  29  906     1449        1 WaitRx
  30  827     1449        1 RxISR(0x360021)
  31  866     1449        1 Stop
  32  764     1449        1 AddrISR(0x360001)
  33  787     1449        1 WaitAddr
  34  764     1449        1 AddrISR(0x360001)
  35  787     1449        1 WaitAddr
humility: ring buffer task_spd::__RINGBUF in spd:
 NDX LINE      GEN    COUNT PAYLOAD
   6  338      700        1 MuxState(Idle, AwaitingSegment)
   7  316      700        1 Rx(0x4a, 0x40)
   8  338      700        1 MuxState(AwaitingSegment, Done)
   9  311      700        4 Initiate(0x1d, false)
  10  311      700        1 Initiate(0x4a, true)
  11  316      700        1 Rx(0x4a, 0x3)
  12  338      700        1 MuxState(Idle, AwaitingSegment)
  13  316      700        1 Rx(0x4a, 0x40)
  14  338      700        1 MuxState(AwaitingSegment, Done)
  15  311      700        4 Initiate(0x19, false)
   0  311      701        1 Initiate(0x4a, true)
   1  316      701        1 Rx(0x4a, 0x3)
   2  338      701        1 MuxState(Idle, AwaitingSegment)
   3  316      701        1 Rx(0x4a, 0x40)
   4  338      701        1 MuxState(AwaitingSegment, Done)
   5  311      701        4 Initiate(0x1b, false)
```
In this instance DIMM 1 is installed but not visible.
[hubris.core.1.gz](https://github.com/oxidecomputer/hubris/files/9274254/hubris.core.1.gz)

refugeesus, 2022-08-10:
OK so we have a lot of mbist runs now and I haven't seen this issue show up and we have good eyes. Link to the data for SN19 is: https://drive.google.com/drive/u/0/folders/1O_Ddk7YUUw9_QaG8vyYGx1i2lAwZHueA and data for two new machines is incoming (takes days to do 100+ runs)

Admittedly someone looking for this error should weigh in on how to induce the problem. We are power cycling between each mbist run (30 second power off time). As we get more data I will continue to upload with eye diagrams.

wesolows, 2022-08-10:
I'm not aware of any way to induce the problem, nor prevent it.  I do know that if one does not drop to A3 or lower, the state (buggy or not) will not change.  If you are dropping into A3/A4 each time, I don't have anything more to offer you.  I did look at one of these archives and agree that you did not encounter the bug.  You also, like me, have 512 GiB installed so that's not a variable either.  We are probably down to software versions and the possibility of a manufacturing defect.

bcantrill, 2022-08-11:
The next time we see this, it would be very helpful to force `humility spd` to read the data (versus the cached data) from both the `mid` bus and the `rear` bus, e.g.:

```console
$ humility spd -b mid
humility: attached to 0483:374e:002F001A4741500520383733 via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   0 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   1 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   2 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   4 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   6 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
$ humility spd -b rear
humility: attached to 0483:374e:002F001A4741500520383733 via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   0 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   1 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   2 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   4 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   6 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
```

wesolows, 2022-09-07:
Took a while to hit this again, but here it is in all its glory:

```
$ humility spd
humility: attached via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   9 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  11 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  13 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
  15 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
wesolows at monolith in ~
$ humility spd -b rear
humility: attached via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   1 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
wesolows at monolith in ~
$ humility spd -b mid
humility: attached via ST-Link V3
ADDR MANUFACTURER              PART                 WEEK YEAR
   1 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   3 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   5 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
   7 Micron Technology         36ASF8G72PZ-3G2E1      43 2021
```
So whatever this is, it happens only when the contents are first cached, or at least it happens then and not reliably every time thereafter.

cbiffle, 2023-04-27:
@wesolows - any sign of this one since the putative fix landed in November?

wesolows, 2023-04-27:
I spent most of the last 3 days rebooting B15 for unrelated reasons, and I did at least occasionally look for this.  Also the 53 boots in my scrollback buffer show no instances of this bug.  Should be safe to close as fixed!

cbiffle, 2023-04-27:
Thanks for checking!

closed 2023-04-27 by cbiffle