oxidecomputer/hubris #669: KSZ8463 dynamic MAC table lies about number of entries

mkeeter, 2022-07-27:
If you read slot 0 in the KSZ8463's MAC table, it will report that there is one entry (strictly, it reports that it's non-empty with "Number of Valid Entries" = 1, so there's an implied +1).

However, if you activate two devices that send packets through it and read entries 0 _and_ 1, they are the two MAC addresses; during each read, it declares that there's only one entry in the table:

```
[entry 0] Ksz8463MacTable(MacTableEntry { count: 0x1, timestamp: 0x0, source: Port1, fid: 0x0, addr: [ 0xe, 0x1d, 0x7f, 0xc3, 0x7, 0x31 ] })
[entry 1] Ksz8463MacTable(MacTableEntry { count: 0x1, timestamp: 0x0, source: Port3, fid: 0x2, addr: [ 0xe, 0x1d, 0x15, 0x70, 0x3d, 0xbb ] })
```

mkeeter, 2022-07-27:
Opened Microchip case #01042036

cbiffle, 2023-04-27:
@mkeeter what work do we need to do on this?

mkeeter, 2023-04-28:
I was holding it wrong; this works fine now.

closed 2023-04-28 by mkeeter