oxidecomputer/hubris #702: Want thermal trip black box data

rmustacc, 2022-08-09:
We recently had a thermal trip event occur because I forgot to install an air-flow shroud. While the system correctly detected that we had a thermal trip and a humility get_state made that very obvious, However, the next question I had was: What was the last temperature we saw?

To answer this I think there are a few things we want to do here in increasing involvement:

* Have some kind of ring buffer of data that we've read so we can denote what the last set of values we saw were. While the last valid measurement is useful, having a few entries that led there will help. This is probably tied into the broader thermal loop restructuring.
* Eventually we want to make sure that we're holding on to data and evacuating it over the management network and transforming it into metrics data (cc @jgallagher, @bnaecker). That may also want to have a different policy of how much data is kept around to cover some amount of network disconnects and other periods were such services are not available. As an example, say we cold start a rack and one system has a thermal trip of the CPU (or really any other component such as a VR), we'd want to keep that around long enough so someone else could contact and discover us and get that information out.

cbiffle, 2022-09-19:
I think there are three things happening here, all of which I'd like to see implemented at some point in the near future.

1. Currently, the `sensors` task lets tasks blow away their last sensor reading by sending `None`. This is bad, and means that any sensor failure typically makes it impossible to get the _previous_ reading. This is pretty high on my things-to-fix list.
2. The `sensors` task will need to start recording enough readings to cover the time between telemetry transmissions, since we don't want to generate network traffic in response to every sensor reading. This will likely mean at least some small circular buffers per sensor, and could be made more ornate by e.g. low-pass filtering or running-averaging sensor readings by channel to decimate the output data. (We will also need corresponding APIs to collect buffered data to prepare a transmission.)
3. Circular buffers are only useful if you were recording the moment the event in question happened. I think we'll wind up wanting a way of defining "events" that, when they occur, snapshot a subset of sensor channels in RAM. Other tasks could be responsible for defining when these events (such as thermtrip) occur and either recording the data locally, or poking the `sensors` task to trigger a snapshot. Later, Humility or the task doing telemetry transmissions would notice that the event had occurred and evacuate the corresponding recorded data.

Recording subsets of sensor data introduces a resource allocation problem: the recorded data needs to be put _somewhere_ that it won't be casually overwritten before it's collected for telemetry. I see three ways of approaching this:

1. **The party detecting the event pays.** If a task (say, gimlet-seq) wants to record a subset of sensor values at an event (say, a power state transition), it is responsible for allocating storage in its RAM and slurping the relevant sensor data into it when necessary. This results in more smaller allocations in many tasks, which helps with fragmentation. However, pulling the sensor data across in response to an event may introduce unacceptable latency, and any recorded data would be lost if the recording task restarted. (The `sensors` task is, in general, not intended to restart, making it a safer place for data.)
2. **The sensors task pays.** At design time, the set of events that may occur and the subsets of data they involve are defined, and the `sensors` task is given static RAM allocations to hold them. Other tasks detecting an event just poke `sensors` with an IPC saying the event has happened, and `sensors` does a local copy of the sensor data into the snapshot storage. This should be significantly faster (since it's a direct in-task copy and fewer IPCs) and guards the snapshot data against restart of driver tasks. However, it puts all snapshot storage into a single place, which as it grows is more likely to cross power-of-two boundaries and lose space to fragmentation.
3. **We best-effort the data into a third party.** When an event occurs, the snapshot data is sent to _another_ task, such as a dedicated `log` task, that records things sent to it in a best-effort circular-buffer fashion. If we send too much, we lose data. This is essentially the approach taken by most log-based and Unix-like systems but would be the last thing I'd reach for in our system: it has all the IPCs/cross-task data copies of option 1 above (and possibly more), introduces a tunable that must be maintained (the amount of RAM dedicated to logs), and will fail in annoying ways under load (by losing important data).

Unless the plan fails to survive contact with reality, I'd go with option 2 (sensor task pays) when implementing black box storage, and possibly rethink it if-and-when fragmentation becomes an issue. I suspect the total amount of snapshot data required is relatively low (it is rarely useful to sample temperatures in our system at >1Hz, say), but, I honestly have no idea at this point.

lzrd, 2022-09-20:
Time stamping, as mentioned above, is important in sensor readings used for thermal management.
For any sensor that is an input to thermal management, it is important to understand how much time it can take for a thermal excursion on each sensor.
If one is able to continuously sample the necessary sensors and that sampling rate is faster than any time to thermal excursion, then go ahead and do that.
If it is not practical or desirable to maintain continuous sampling, then it is important to deepen one's understanding of the system in order to properly tune sampling.
For example, the SP3 has the ability to generate quite a bit of heat. Without its cooling solution, it might reach critical temperatures faster then the best thermal management reaction time resulting in an automatic shutdown and perhaps some damage. With a properly installed heat sink, working fans, and an un-blocked source of sufficiently cool and sufficiently dense air, the worst case thermal excursion time from a nominal temperature reading to a thermal excursion may be over a minute (or some other time longer than you expected). It might be reasonable to lengthen time between samples on the SP3.

Time stamps on sensor data allow determination of freshness. e.g. "This data not considered valid after uptime=123456".

The IPMI System Event Log is a useful reference as far as black box type logging that has been important in other products.
IPMI is a deprecated exercise in standards accretion including vendor escapes.

See https://www.intel.com/content/dam/support/us/en/documents/server-products/SEL_TroubleshootingGuide.pdf
and also click through access to the IPMI spec (note the deprecation warning there.) https://www.intel.com/content/www/us/en/products/docs/servers/ipmi/ipmi-home.html