oxidecomputer/hubris #726: humility qspi failed: request failed: Some(Fault(ReturnStackOverflow))

rmustacc, 2022-08-23:
While trying to use humility to write to the qspi using -W, we end up failing with a return stack overflow:

```
rm@cadbury ~ $ pfexec /gimlet/bin/humility-0.8.8 -t sn9 qspi -W ./milan-gimlet-b.img 
humility: attached to 0483:374e:002F001A4741500520383733 via ST-Link V3
humility: erasing 33554432 bytes...
humility qspi failed: request failed: Some(Fault(ReturnStackOverflow))
rm@cadbury ~ $ pfexec /gimlet/bin/humility-0.8.8 -t sn9 ringbuf
humility: attached to 0483:374e:002F001A4741500520383733 via ST-Link V3
humility: ring buffer drv_gimlet_seq_server::__RINGBUF in gimlet_seq:
 NDX LINE      GEN    COUNT PAYLOAD
  56  437    62383        1 GetState
  57  364    62383        1 Status(0x0, 0x8, 0xff)
  58  437    62383        1 GetState
  59  364    62383        9 Status(0x0, 0x8, 0xff)
  60  437    62383        1 GetState
  61  364    62383        1 Status(0x0, 0x8, 0xff)
  62  437    62383        1 GetState
  63  364    62383        9 Status(0x0, 0x8, 0xff)
   0  437    62384        1 GetState
   1  364    62384        1 Status(0x0, 0x8, 0xff)
   2  437    62384        1 GetState
   3  364    62384        9 Status(0x0, 0x8, 0xff)
   4  437    62384        1 GetState
   5  364    62384        2 Status(0x0, 0x8, 0xff)
   6  437    62384        1 GetState
   7  364    62384        8 Status(0x0, 0x8, 0xff)
   8  437    62384        1 GetState
   9  364    62384        2 Status(0x0, 0x8, 0xff)
  10  437    62384        1 GetState
  11  364    62384        8 Status(0x0, 0x8, 0xff)
  12  437    62384        1 GetState
  13  364    62384        2 Status(0x0, 0x8, 0xff)
  14  437    62384        1 GetState
  15  364    62384        8 Status(0x0, 0x8, 0xff)
  16  437    62384        1 GetState
  17  364    62384        2 Status(0x0, 0x8, 0xff)
  18  437    62384        1 GetState
  19  364    62384        8 Status(0x0, 0x8, 0xff)
  20  437    62384        1 GetState
  21  364    62384        3 Status(0x0, 0x8, 0xff)
  22  437    62384        1 GetState
  23  364    62384        7 Status(0x0, 0x8, 0xff)
  24  437    62384        1 GetState
  25  364    62384        3 Status(0x0, 0x8, 0xff)
  26  437    62384        1 GetState
  27  364    62384        7 Status(0x0, 0x8, 0xff)
  28  437    62384        1 GetState
  29  364    62384        3 Status(0x0, 0x8, 0xff)
  30  437    62384        1 GetState
  31  364    62384        7 Status(0x0, 0x8, 0xff)
  32  437    62384        1 GetState
  33  364    62384        4 Status(0x0, 0x8, 0xff)
  34  437    62384        1 GetState
  35  364    62384        6 Status(0x0, 0x8, 0xff)
  36  437    62384        1 GetState
  37  364    62384        4 Status(0x0, 0x8, 0xff)
  38  437    62384        1 GetState
  39  364    62384        6 Status(0x0, 0x8, 0xff)
  40  437    62384        1 GetState
  41  364    62384        4 Status(0x0, 0x8, 0xff)
  42  437    62384        1 GetState
  43  364    62384        6 Status(0x0, 0x8, 0xff)
  44  437    62384        1 GetState
  45  364    62384        5 Status(0x0, 0x8, 0xff)
  46  437    62384        1 GetState
  47  364    62384        5 Status(0x0, 0x8, 0xff)
  48  437    62384        1 GetState
  49  364    62384        5 Status(0x0, 0x8, 0xff)
  50  437    62384        1 GetState
[... 378 lines cut ...]
  27   38        1        1 Validate(0x18)
  28   38        1        1 Validate(0x19)
  29   38        1        1 Validate(0x1a)
  30   38        1        1 Validate(0x1b)
  31   38        1        1 Validate(0x1c)
  32   38        1        1 Validate(0x1d)
  33   38        1        1 Validate(0x1e)
  34   38        1        1 Validate(0x1f)
  35   38        1        1 Validate(0x20)
  36   38        1        1 Validate(0x21)
  37   38        1        1 Validate(0x22)
  38   38        1        1 Validate(0x23)
  39   38        1        1 Validate(0x24)
  40   38        1        1 Validate(0x25)
  41   38        1        1 Validate(0x26)
  42   38        1        1 Validate(0x27)
  43   38        1        1 Validate(0x28)
  44   38        1        1 Validate(0x29)
humility: ring buffer vsc85xx::__RINGBUF in net:
 NDX LINE      GEN    COUNT PAYLOAD
   0   26        1        1 Vsc8562InitSgmii(0x1e)
   1   22        1        1 ViperPatch(0x1e)
   2  133        1        1 GotCrc(0x336a)
   3  133        1        1 GotCrc(0xfb48)
```

bcantrill, 2022-08-24:
So, as it turns out, this is because the host flash was mux'd to the CPU (that is, we were attempting to flash while not in A2). The failure mode here is clearly awful, which is (broadly) addressed by https://github.com/oxidecomputer/humility/pull/220 -- but to make this even crisper, we should be failing immediately when we know we're in the wrong mux state; that is covered under https://github.com/oxidecomputer/hubris/pull/730.

cbiffle, 2023-04-27:
As both of the fixes @bcantrill mentioned above have merged, I'm going to close this -- I don't believe we've seen it since.

closed 2023-04-27 by cbiffle