oxidecomputer/hubris #731: get address for DICE artifacts from config

flihp, 2022-08-24:
Our prototype currently hard codes these addresses in the source code. Getting them from the `chip.toml` at build time would be better. At first look doing so would require adding some plumbing to xtask. The data we would need is in the 'chip' key from the top level toml table but AFAIK the currently available methods for getting at the app toml exposed to tasks at build time aren't sufficient:
- `build_util::config` exposes the app config that includes tables under `[config]`: https://github.com/oxidecomputer/hubris/blob/09fb9271df0208dc571ade28feb0503c78707817/build/util/src/lib.rs#L50
- `build_util::task_config` exposes tables under the `[task_name]` table: https://github.com/oxidecomputer/hubris/blob/09fb9271df0208dc571ade28feb0503c78707817/build/util/src/lib.rs#L57

cbiffle, 2023-04-27:
It seems like there are symbols being generated by the build system that can give you the address of this area. Is this still a thing?

flihp, 2023-05-08:
This work has fallen behind other tasks and I haven't thought much about it beyond the initial comment. AFAIK this is mostly a plumbing exercise to surface the required data at build time.