oxidecomputer/hubris #736: improve RoT stage0 FWID calculation

flihp, 2022-08-29:
The current implementation is naive as it relies on untrusted data to determine the size of the image being hashed. With an attacker controlling this data we could be tricked into hashing less than the whole image. This would be evident in the FWID included in the Alias cert so this scenario isn't particularly scary. What would be interesting though is if an attacker found an existing vulnerability in the hubris image that could be used to control execution flow. If the attacker were also able to modify flash in the RoT, then they could persist code in the unused portion of the last flash page. They could then use their ability to manipulate execution to run the persisted payload and the existing method of calculating the FWID would not include the attackers code.

The FWID calculation should cover all flash pages allocated in the A / B image slot being booted.

lzrd, 2023-07-24:
The hash of the declared image is important to drive update.
The hash of all programmed pages in an image slot is interesting from a security perspective.
If we were to implement some confidential storage, we might want to know that some blocks were programmed rather than a hash which may leak information.
The count of all programmed pages in an image slot would be a safe and thing to implement in the near term.

flihp, 2023-08-16:
I've got a wip for this here: https://github.com/oxidecomputer/hubris/tree/xtask-rot-fwid. I've tried to link this branch to this issue using the 'development' thing in the right hand pane but it can't find the branch. This was pushed a few minutes ago so maybe it just takes time to propagate to the UI? I'll update that if it fixes itself.

closed 2023-08-25 by flihp