oxidecomputer/hubris #741: mutable_statics!() can put large data structures on the stack

jgallagher, 2022-08-30:
I ran into this trying to put a more-complex-than-plain-arrays type inside a `mutable_statics!()` call and noticed my `stackmargin` had jumped up by a few kilobytes. I made a small reproducer in this commit: https://github.com/oxidecomputer/hubris/commit/7910bd1d457f6c140400eda452b26aab25fdf4c5

On that commit, `humility stackmargin` reports

```
ID TASK                STACKBASE  STACKSIZE   MAXDEPTH     MARGIN
10 udpecho            0x20002000       4096       2624       1472
```

This definitely smells like compiler optimizer issues; if you apply this diff to that commit:

```diff
--- a/task/udpecho/src/main.rs
+++ b/task/udpecho/src/main.rs
@@ -64,7 +64,7 @@ fn main() -> ! {
 const STACK_SMASHER_BUF_SIZE: usize = 1024;

 struct StackSmasher {
-    counter1: Option<usize>,
+    counter1: usize,
     counter2: usize,
     data: [u8; STACK_SMASHER_BUF_SIZE],
 }
@@ -72,7 +72,7 @@ struct StackSmasher {
 impl Default for StackSmasher {
     fn default() -> Self {
         Self {
-            counter1: None,
+            counter1: 0,
             counter2: 0,
             data: [0; STACK_SMASHER_BUF_SIZE],
         }
```

the stack usage drops back down to the much more reasonable level it is without any of these changes:

```
ID TASK                STACKBASE  STACKSIZE   MAXDEPTH     MARGIN
10 udpecho            0x20002000       4096        656       3440
```

In the disassembly of the commit with the high stack usage, I see a couple large memcpys that aren't there in the lower stack usage variant:

```asm
0000007c <main>:
      7c:       b580            push    {r7, lr}
      7e:       466f            mov     r7, sp
      80:       f5ad 6d1b       sub.w   sp, sp, #2480   ; 0x9b0
      84:       f8df 0718       ldr.w   r0, [pc, #1816] ; 7a0 <main+0x724>
      88:       f240 31fe       movw    r1, #1022       ; 0x3fe
      8c:       8800            ldrh    r0, [r0, #0]
      8e:       4288            cmp     r0, r1
      90:       d105            bne.n   9e <main+0x22>
      92:       f8df 0758       ldr.w   r0, [pc, #1880] ; 7ec <main+0x770>
      96:       212c            movs    r1, #44 ; 0x2c
      98:       f8df 2754       ldr.w   r2, [pc, #1876] ; 7f0 <main+0x774>
      9c:       e35d            b.n     75a <main+0x6de>
      9e:       f36f 209f       bfc     r0, #10, #22
      a2:       f7ff fffe       bl      18b0 <_ZN7userlib24sys_refresh_task_id_stub17h0b44fa0a3d65c6fcE>
      a6:       f8df 86fc       ldr.w   r8, [pc, #1788] ; 7a4 <main+0x728>
      aa:       2301            movs    r3, #1
      ac:       e8d8 1f4f       ldrexb  r1, [r8]
      b0:       e8c8 3f42       strexb  r2, r3, [r8]
      b4:       2a00            cmp     r2, #0
      b6:       d1f9            bne.n   ac <main+0x30>
      b8:       bb29            cbnz    r1, 106 <main+0x8a>
      ba:       f50d 6a8b       add.w   sl, sp, #1112   ; 0x458
      be:       9012            str     r0, [sp, #72]   ; 0x48
      c0:       f10a 0404       add.w   r4, sl, #4
      c4:       f8df 56e8       ldr.w   r5, [pc, #1768] ; 7b0 <main+0x734>
      c8:       f240 460c       movw    r6, #1036       ; 0x40c
      cc:       f10d 0b50       add.w   fp, sp, #80     ; 0x50
      d0:       f04f 0900       mov.w   r9, #0
      d4:       b1e6            cbz     r6, 110 <main+0x94>
      d6:       4620            mov     r0, r4
      d8:       f240 4104       movw    r1, #1028       ; 0x404
      dc:       f7ff fffe       bl      192c <__aeabi_memclr4>
      e0:       4658            mov     r0, fp
      e2:       4651            mov     r1, sl
      e4:       f44f 6281       mov.w   r2, #1032       ; 0x408
      e8:       f7ff fffe       bl      1c3c <__aeabi_memcpy4>
      ec:       4628            mov     r0, r5
      ee:       4659            mov     r1, fp
      f0:       f840 9b04       str.w   r9, [r0], #4
      f4:       f44f 6281       mov.w   r2, #1032       ; 0x408
      f8:       f7ff fffe       bl      1c3c <__aeabi_memcpy4>
```

I did not dig any further; the clue to simplify the type I was giving to `mutable_statics!()` was enough to fix my real issue (for now?).

labbott, 2022-08-30:
A couple of ideas to maybe narrow something down: 1) `usize` is going to take up less structure space than `Option<usize>`, if you add something in that field that's the exact same size do you see the same behavior? 2) Does changing the size of `STACK_SMASH_BUFFER` affect the behavior at all?

jgallagher, 2022-08-31:
> 1) `usize` is going to take up less structure space than `Option<usize>`, if you add something in that field that's the exact same size do you see the same behavior?

If I replace the `Option<usize>` with these two fields:

```diff
 struct StackSmasher {
-    counter1: Option<usize>,
+    flag: bool,
+    counter1: usize,
     counter2: usize,
```

I get a struct that's the same size as it was with the `Option<usize>`, but the stack usage stays small:

```
ID TASK                STACKBASE  STACKSIZE   MAXDEPTH     MARGIN
10 udpecho            0x20002000       4096        656       3440
```

> 2) Does changing the size of `STACK_SMASH_BUFFER` affect the behavior at all?

It doesn't appear to. Dropping `STACK_SMASH_BUFFER` to 256 reduces the stack usage by 1536 (exactly 2x the reduction in size, implying we're getting two copies on the stack?):

```
ID TASK                STACKBASE  STACKSIZE   MAXDEPTH     MARGIN
10 udpecho            0x20002000       4096       1088       3008
```

Increasing `STACK_SMASH_BUFFER` to 1200 increases the stack usage by 352 (exactly 2x the increase in size):

```
ID TASK                STACKBASE  STACKSIZE   MAXDEPTH     MARGIN
10 udpecho            0x20002000       4096       2976       1120
```

hudson-ayers, 2022-10-26:
I just stumbled across this while perusing the Hubris Github, but have run into a similar problem before in Tock. The `None` vs. `0` difference may be because `None` is not guaranteed to be represented as 0 in the binary, so the constructor for that type goes from, basically, "write all 0s into the space allocated for this object" to "copy this 1000 byte ASCII string over from .data", even though that 1000 byte string may be all 0s and a single 1. 

The next issue is that Rust does not have any support for "placement new" to initialize statics in place, so rather than directly initializing the static, it first runs the constructor (allocating space for the return value on the stack), then copies that allocated object into the space for the static. I am not sure why you are seeing double the size change just on the stack though, it seems you should see the extra space used once on the stack and once in `.data` or wherever static initializers are placed.