oxidecomputer/hubris #752: set notBefore in DICE certs

flihp, 2022-09-02:
Currently the cert templates have this set to the time that they were created. When device id certs are created off-platform this will result in decedents having notBefore values that pre-date their parent. This is weird and instead they should either use the same time or a later one. For self signed device id certs we could generate this at build time but we must be sure that we don't break reproducible builds (timestamps are the worst).

flihp, 2022-09-02:
Before we spend time on this we should probably understand what we're using the field for in cert validation. AFAIK we probably won't?

kc8apf, 2022-09-02:
Seems unlikely.  We'll need to use the cert in plenty of circumstances
where we won't know UTC so we'll just need to trust that the cert is valid
anyway.

closed 2022-12-05 by flihp