oxidecomputer/hubris #766: Running Hubris with a runtime offset

mchodzikiewicz, 2022-09-10:
Hi, 
I have posted an initial question in the discussion section but it seems to be dead (https://github.com/oxidecomputer/hubris/discussions/744).

I would be super happy if you could reply in few words - at least if what I write here makes any sense - am I missing something that already allows such a use case or something that is a showstopper here?

I see a huge value for Hubris in the market I come from (IoT devices for healthcare), but in that case the devices are standalone and firmware update often needs to be possible within the same application without any external probe or dongle.
Playing around with Hubris is already a great escape from everyday's work for me so maybe I might do something valuable with it or at least give an idea to somebody else.

Merge viable solution I imagine is a bootloader for Hubris with support from dedicated task:
* bootloader toml file with partition layout and perhaps additional configs for the keys to check if the firmware is properly signed etc
* bootloader that's flashed at the beginning of the flash that decides which partition to boot from
* task that can:
  * flash a new image into the unactive partition and overwrite the bootloader header to try boot from the other partition
  * receive acknowledgement that the new image booted properly and overwrite the bootloader header to make newly booted image a permanent one

There's a long way to that point, but it would enable using Hubris for standalone IoT devices that requires FOTA capabilities, so I want to at least give it a try with a simple POC

I started to investigate what's necessary to make Hubris running from runtime configurable flash offset.
What I already know is that I need to:
* Enable PIC for kernel and tasks
* Pass an image flash offset from a bootloader to `main()` and then to `start_kernel()` and add it to some kernel struct
* Use the flash offset for the `reinitialize()` and `apply_memory_protection()` and most likely few other places i have not yet spotted
* Make Humility read the runtime image offset and apply for all addr to symbol translation (this gives me a headache, but might be not that bad)

I feel like for the initial work, reusing stage0 for a dummy bootloader might be a good idea since as far as I can see it is included in a Hubris archive and flashed, so less friction from tooling I guess... 
It should also be fairly simple to trick Humility to flash bootloader without an offset and the rest with offset applied by additional parametrization of `generate_ihex_from_elf()`

PS. I understand that you are extremely busy making your first product happen but unfortunately you published the source code and it looks very interesting and partially undocumented, therefore my questions :smile: all the best for the team and thanks for the podcasts!

kc8apf, 2022-09-19:
When we've tried to use PIC for tasks (to allow using the same built task in different applications), we've run into various problems with ROPI/RWPI.  In a microcontroller, there is very little RAM so code typically executes directly from flash with only runtime data in RAM.  That means there isn't a single linear offset for the task as the read-only code portions will be relocated independently of the read-write RAM portions.  Not only does this mean the Hubris kernel needs to handle the offsets but also that the task code must be built to use two different offsets.  That's what ROPI/RWPI is and it is an upstream LLVM feature.  We haven't poked at it recently to see what has progressed but it definitely wasn't working when we last tried.

mchodzikiewicz, 2022-09-19:
Ah, ok - It feels obvious now and I feel dumb, but I thought that RAM segments will have a zero offset (because they do in PIC enabled apps with non-mem-isolated scenario I am used to...)

If you have a stale side branch or something I could use a starting point, I'd gladly volunteer to revise if it is any better now

mchodzikiewicz, 2022-09-20:
Hmm - I would still like to check ROPI/RWPI just because it is interesting and that portable precompiled tasks are a valuable thing to have, however - in the usecase I am talking about, the RAM offset is indeed constant since RW segments will always be initialized in the same RAM addresses.
I believe it is only about where the `.data` of a final binary merged from all task's elf files lies in flash so it is copied to RAM during a startup, and this is I believe a default behavior when using `-fPIC` in an ordinary C/C++ `arm-none-eabi-*` binary with a typical startup file.
In case of Rust I believe this is already correctly handled by `cortex-m-rt` crate, but I need to dig a bit more to be sure.

kc8apf, 2022-09-20:
`-fPIC` without ROPI/RWPI assumes that everything will be copied into RAM and thus all symbols are affected by the offset.  Even if the RAM offset should be 0, the symbols needs to be divided into two groups so they can be adjusted independently.  Otherwise, you'll have no way of knowing which symbols need to have the offset applied and which do not.

mchodzikiewicz, 2022-09-21:
I knew it works like this for binaries compiled for hosted environments but I kinda thought that for bare metal targets `-fPIC` works like ROPI-only but apparently it is more complex and compiler specific - I had an interesting read down the rabbit hole - thanks :slightly_smiling_face: 

I guess I'll first try to achieve my intended behavior with a minimal working rust app instead of trying to do this with such a complex build system Hubris has