oxidecomputer/hubris #778: use const generics in DICE cert trait

flihp, 2022-09-14:
This interface would ideally return sized arrays for the generic 'get_range' function. This isn't yet possible w/o using unstable features:
```rust
trait Foo {
    const BAR: usize;
    
    fn baz(self) -> [u8; Self::BAR];
}
```
Currently this produces an error:
```
error: generic parameters may not be used in const operations
 --> src/lib.rs:6:26
  |
6 |     fn baz(self) -> [u8; Self::BAR];
  |                          ^^^^^^^^^ cannot perform const operation using `Self`
  |
  = note: type parameters may not be used in const expressions
  = help: use `#![feature(generic_const_exprs)]` to allow generic const expressions
```

flihp, 2022-09-14:
upstream tracking issue for the related feature: https://github.com/rust-lang/rust/issues/76560

cbiffle, 2023-04-27:
So, fwiw, you can fake this.

```rust
trait Foo {
    type ARRAY: AsMut<[u8]> + AsRef<[u8]>;

    fn baz(self) -> Self::ARRAY;
}
```

However, consider whether returning an array is what you want, since as the array gets larger, the cost of a potential compiler-generated copy increases. While it's less clean it's sometimes better for code generation to do

```rust
    fn baz(self, out: &mut Self::ARRAY);
```