oxidecomputer/hubris #829: Jefe state notification not being sent

mkeeter, 2022-10-06:
This is a weird one.

On Gimlet SN22, after flashing it with acdd7a91972cbe34f25778fd5035a60d5fe24313, I noticed that the `net` task was hanging during [`preinit`](https://github.com/oxidecomputer/hubris/blob/acdd7a91972cbe34f25778fd5035a60d5fe24313/task/net/src/bsp/gimlet_b.rs#L50-L77).

This indicates that Jefe had not sent a notification when it went to `PowerState::A2`, which was confirmed by not seeing `JefeNotification { now: 0x4c9, state: A2 }` in the ringbuf for `host_sp_comms`.

When asked through `hiffy -c Jefe.get_state`, `jefe`, properly reported that it was in state `0x01`, i.e. A2.

When I sent the system to A0, the net task remained wedged, despite the state changing to `0x05` (as reported by `Jefe.get_state`).

Running `humility reset` also failed to unwedge the system.  Reflashing, however, caused the system to come up normally.

Here's the dump: [hubris.core.jefe-state-stuck.zip](https://github.com/oxidecomputer/hubris/files/9726522/hubris.core.jefe-state-stuck.zip)

cbiffle, 2022-10-06:
In that core, I can confirm that the `net` task has not received an unhandled notification (`tasks -v` shows its `notifications` word is zero).

I have extracted the jefe binary using `humility -d your-core-file extract elf/task/jefe -o /tmp/jefe` and then disassembled it using `arm-none-eabi-objdump --demangle -d /tmp/jefe | less` and reviewed it for `svc` instructions that indicate syscalls. I see several in our syscall stubs... but I do _not_ see one doing a `sys_post` call.

This suggests to me that Jefe in this archive was built from an `app.toml` that had an empty `on-state-change` config, and Jefe is not sending notifications. Why that's in your gimlet archive is beyond me -- the config for gimlet is correct.

When you say

>  Reflashing, however, caused the system to come up normally.

it sounds like you mean _rebuilding,_ (which you've confirmed in chat,) which may have undid whatever build weirdness you were getting. In particular, if this archive was built with `--dirty` in a source tree where you'd previously built a _different_ app without an intervening clean, you may have picked up the wrong jefe.

cbiffle, 2022-10-06:
Analysis in chat noted that if you had somehow managed to build a gimlet archive using _sidecar's_ build of Jefe, you would get these results. Not clear how that could have happened though.

mkeeter, 2022-10-07:
This is likely due to #836  and rampant use if the `--dirty` flag

closed 2022-10-07 by mkeeter