oxidecomputer/hubris #835: map memory into hubris tasks read-only

flihp, 2022-10-06:
We're currently mapping DICE artifacts into task memory using the mechanism from chips.toml intended for mapping in memory mapped device registers (https://rfd.shared.oxide.computer/rfd/0282#_ram). AFAIK the current implementation maps all regions as read-write which makes perfect sense for the original use-case. A currently open PR for the DICE code moves all common certs (DeviceId & the intermediate CA cert) into a single memory range as a means to save stack space in stage0. Tasks receiving these certs should never change them and now that they're in a location that will be accessed by multiple tasks they need to be read-only. The most likely approach is to add a new field to the toml where read-only / read-write can be recorded. This data makes it to the Hubris kernel as serialized ron via the `HUBRIS_KCONFIG` environment variable. The kernel would then need to extract this data to configure the mapping appropriately. read only mappings seem possible according to: https://developer.arm.com/documentation/dui0552/a/cortex-m3-peripherals/optional-memory-protection-unit/mpu-access-permission-attributes

cbiffle, 2022-10-13:
There've been a couple previous cases where folks abused the peripherals mechanism to do a read-only shared mapping -- I definitely agree that we need this.

flihp, 2023-02-09:
There's a draft PR up adding an attestation task to hubris that achieves a read-only mapping for the common parts of the DICE cert chain using the 'sections' / memory.toml mechanism. This allows us to make the memory read only, but prevents us from mapping the memory region into multiple tasks. Since the use-case for mapping this memory into multiple tasks isn't an MVP requirement the current approach is sufficient.

cbiffle, 2023-04-27:
@mkeeter does the more general memory region support you added let us fix this issue with the DICE region?

mkeeter, 2023-04-27:
I think we should be using the new `extern-regions` for the data handed from pre-Hubris into Hubris (which I believe is `dice_certs` in the PR), since that's magical memory that Rust doesn't know about.  This region can be configured as read-only, which should be enforced by the kernel.

We can use `memory.toml` to carve out the `dice_alias` region, since that's R/W and written to by Hubris (I think!).

In both cases, I believe we can map the same memory into multiple tasks, so I'm a little confused by @flihp's comments above.

closed 2023-05-01 by flihp