oxidecomputer/hubris #873: Power shelf sensor support

mkeeter, 2022-10-24:
Once we can take inventory of the power shelves (#872), we should monitor voltage / current / fan speed / temperature (?) and log it to the `sensors` task.

isobering, 2022-10-25:
Here's the link to Murata's application note with PMBus addresses for the EEPROM and microcontroller in each PSU:
https://www.murata.com/-/media/webrenewal/products/power/appnote/acan-114.ashx

isobering, 2022-10-25:
Here's the list of things we'd like to monitor:
- `READ_VIN` (command code 88h, page 6)
- `READ_VOUT` (command code 8Bh, page 6)
- `READ_IIN` (command code 89h, bage 6)
- `READ_IOUT` (command code 8Ch, page 6)
- `READ_TEMPERATURE_1` (command code 8Dh, page 7)
- `READ_TEMPERATURE_2` (command code 8Eh, page 7)
- `READ_TEMPERATURE_3` (command code 8Fh, page 7)
- `READ_FAN_SPEED_1` (command code 90h, page 7)
- `READ_FAN_SPEED_2` (command code 91h, page 7)

Logging input voltage/current and output voltage/current will let us perform basic rack-level power monitoring (one of the PSC's original functional requirements). Logging PSU temperature and fan speed will let us demonstrate some basic state-of-health monitoring (another one of the PSC's original functional requirements).

PR #885 (2022-10-28): Integrate Murata power shelf sensors
This PR adds communication to the actual Murata power shelves!

It includes a few pieces:

- Modifications to the build system to support sensors that are decoupled from rails.  This is analogous to https://github.com/oxidecomputer/humility/pull/252 , and requires Humility 0.9.3 (enforced by a Hubris build archive bump)
- I2C drivers for the M24C02 EEPROM (used as a FRU ID in the power shelf) and the MWOCP68 itself
- Support for the power shelves in the `power` task, which means they work with `humility pmbus` (they were already in the `pmbus` crate, thanks [... cut]
commit 0794064e (2022-10-28): Integrate Murata power shelf sensors (#885)

This PR adds communication to the actual Murata power shelves!

It includes a few pieces:

- Modifications to the build system to support sensors that are decoupled from
  rails.  This is analogous to humility#252 , and requires Humility 0.9.3
  (enforced by a Hubris build archive bump)
- I2C drivers for the M24C02 EEPROM (used as a FRU ID in the power shelf) and
  the MWOCP68 itself
- Support for the power shelves in the `power` task, which means they work with
  `humility pmbus` (they were already in the `pmbus` crate, thanks @bcantrill!)
- A new  [... cut]
closed 2022-10-28 by mkeeter