oxidecomputer/hubris #890: host-sp-comms set_status_impl() not taking power domain into account

rmustacc, 2022-10-29:
The `host-sp-comms` tasks understandably exists across power domains. In particular, it can exist in A2 as it is part of the host normal path for executing commands like a reboot and related. However, this puts us in a particularly gnarly power domain case where we can possibly cross power domains. Normally the UART is protected by a buffer for the actual data and ready to send style pins that are treated as push-pull. However, the interrupt pin that we're using is not protected by that buffer. Unfortunately, all of the UART pins should only be driven from the ACPI state S0 according to the docs, which we treat as A0.

Because the task can operate across domains the current construct in `set_status_impl()` is liable to cause back driving. UART backriving has been a serious concern and we've proven issues on Ethanol-X. While we do have isolation between the main UART signals, we didn't for the SP->SP3 interrupt line because at the time use was less clear and we figured the SP could do the right thing semantically.

There are two approaches to this to consider:

1. `set_status_impl()` needs to be aware of the power domain and if we are transitioning away from A0 at all (more specifically the sequencer would toggle the uart enable lines) then we need to disable output on the GPIO.
2. Alternatively, because there is a pull up resistor on this, we may be able to get away with a bit of cheating and treating this like an open-drain pin on the SP side. Technically this is a push-pull from the SP3 side, but depending on if the interrupt is edge or level triggered and the rise times of this, that would allow us to potentially pretend the power domain doesn't exist as if the line is driven to 3.3V the pull up will cover it if it's in A0 and will not if we're not in A0 as the pull is in that domain.

cbiffle, 2022-10-29:
I'd argue for treating it as open drain if the pullup is to the correct power domain. That change is super easy on the SP side.

citrus-it, 2022-10-30:
I've done some testing today with the SP side set to `OpenDrain` and no pullup/down configured, and the Host side set to input/8KPU, and I haven't been able to reproduce the problem in https://github.com/oxidecomputer/stlouis/issues/189
The interrupt handling code on the Host side is also still working as I'd expect (although it's currently doing polling, it sees the interrupt line go down and up as the host-sp-comms status value is changed from and back to 0).

PR #891 (2022-10-31): host-sp-comms interrupt line to SP3 should be open drain
Here's a proposed PR to fix #890 by configuring the SP-to-host GPIO as open drain.
closed 2022-10-31 by citrus-it