oxidecomputer/hubris #908: Want SPs to automatically go to A0

rmustacc, 2022-11-03:
As part of both the product and compliance testing we need an SP to turn everything into A0 automatically.

jclulow, 2022-11-03:
Hopefully this can be configurable at build time somehow?  I'd like lab machines and the manufacturing SP image to be able to remain at A2 until directed otherwise if we can.

arjenroodselaar, 2022-11-08:
Changing this is a one-liner, but indeed there is value in having some images stay in A2 while prod images transition to A0 whenever they can. We should think about how we can achieve this.

arjenroodselaar, 2022-11-09:
We can certainly do this using a separate configuration file and feature flags. We should think about how to keep those configurations in sync though, because with several hardware revisions and two configurations its easy for these configs to drift apart if we have to maintain them all by hand.

mkeeter, 2022-11-14:
Talking about this in the Hubris Huddle, we have lots of options:

- Build separate images for Lab versus Rack
    - Requires hand-maintenance, doubles image count for Gimlet and Sidecar
    - With hypothetical "TOML include", maintenance load is reduced
- Detect and change behavior at runtime, using a single image
    - Using a key in the FRU EEPROM?
    - Using data in the voltage regulators non-volatile scratch memory?
    - Add a "suppress boot" message to the MGS protocol, then have lab machines broadcast this message at ~1 Hz and wait ~2 seconds before autobooting to A0?

Taking a step back, we're wondering _why_ it's important to avoid having _lab_ machines go to A0.

If this is mostly important for manufacturing, we could make sure we don't go to A0 without the voltage regulators correctly programmed; in fact, we should do this regardless of autobooting mode.

jclulow, 2022-11-15:
> Taking a step back, we're wondering _why_ it's important to avoid having _lab_ machines go to A0.

Mostly it is about control, and not having unexpected things happen.  Generally lab machines are being used for exploring behaviours outside the norm.  Right now, I can write a new SP image and poke at it to determine if it is doing what I want it to do without it booting up the host automatically each time I do that.

There are also long periods where the machines sit idle; e.g., while people are asleep.  There is no UPS in the lab in Emeryville, and people also occasionally (by accident) interrupt power or USB connections while things are being moved around physically.  We also sometimes update the software on the control PCs and reboot them in the middle of the night.  It would be good for those hosts not to just boot up automatically if the SP is reset due to incidental stuff going on.

Looking at the options you've mentioned:

---

>     * Detect and change behavior at runtime, using a single image

At face value, that seems fine to me, modulo these concerns:

>       * Using a key in the FRU EEPROM?

What is the write cycle limit on FRU EEPROMs?  What about machines that come back from the field (RMA, etc), where their FRU EEPROM was previously permanently write-locked; does this mean we would be unable to (through reprogramming) end up with the machine in a lab configuration for whatever unimagined future purpose?

>       * Using data in the voltage regulators non-volatile scratch memory?

What is the write cycle limit for this scratch space?  I noticed when I was programming the regulators on a blank Gimlet that it seemed to suggest there was a very limited quantity of "slots" for writes.  I'm not sure if this is the scratch memory, or if that's separate?

>       * Add a "suppress boot" message to the MGS protocol, then have lab machines broadcast this message at ~1 Hz and wait ~2 seconds before autobooting to A0?

I don't like this one, chiefly because if for some reason we stop broadcasting the hold signal (software error, control PC failure, people touching things in the lab failure) then we end up pretty quickly back in the "things booting up that I did not want to boot up" place.

---

>     * Build separate images for Lab versus Rack
>       * Requires hand-maintenance, doubles image count for Gimlet and Sidecar
>       * With hypothetical "TOML include", maintenance load is reduced

Aside from maintenance issues with the current application description format, this one seems best for lab users?  Unless I'm misunderstanding the write cycle limit on the flash for the SP.

One would presumably be able to tell from `humility manifest` what kind of image is in play, and to change the behaviour you just have to `humility flash`.  This also makes it pretty easy to understand when the machine is going to power up on its own or not in the manufacturing case as well; it would only happen after the final step where we've programmed the production image into the machine.

> If this is mostly important for manufacturing, we could make sure we don't go to A0 without the voltage regulators correctly programmed; in fact, we should do this regardless of autobooting mode.

I agree that's probably a good safety measure, but I don't think it is, on its own, sufficient for manufacturing.  In particular, we'll need to program the voltage regulators early in the process, because some of the manufacturing steps will depend on executing various host OS images (e.g., to program the M.2 devices inside the chassis).

jclulow, 2022-11-26:
I just wanted to highlight that we need to move on this pretty promptly, as it's needed for compliance which is happening in a minute.

mkeeter, 2022-11-27:
Roger, I'll take a look on Monday.

closed 2022-11-29 by mkeeter