oxidecomputer/hubris #910: I2C hang on STM32H7

mkeeter, 2022-11-03:
On `igor`'s Gimlet (`sn5`), we noticed that the `i2c_driver` task was stuck while waiting for a response from I2C4.

This brought down the `thermal` and `power` tasks, and is generally Bad News for the system.

Here's the relevant section of the ringbuf:
```
humility: ring buffer drv_stm32xx_i2c::__RINGBUF in i2c_driver:
 NDX LINE      GEN    COUNT PAYLOAD
  ...
  10  426    44830        1 WaitISR(0x21)
  11  516    44830        1 WriteISR(0x21)
  12  516    44830        2 WriteISR(0x8021)
  13  516    44830        1 WriteISR(0x8023)
  14  546    44830        3 WriteWaitISR(0x8021)
  15  546    44830        1 WriteWaitISR(0x8061)
  16  608    44830        2 ReadISR(0x8021)
  17  608    44830        1 ReadISR(0x8025)
  18  608    44830        2 ReadISR(0x8021)
  19  608    44830        1 ReadISR(0x8025)
  20  643    44830        1 ReadWaitISR(0x8021)
  21  643    44830        1 ReadWaitISR(0x1061)
  22  426    44830        1 WaitISR(0x1021)
  23  516    44830        1 WriteISR(0x21)
  24  516    44830        2 WriteISR(0x8021)
```

`NDX` 10-21 are a successful read; 22-24 are the failed read getting stuck.

It's waiting for `TXDIS` to go high, which never happens, so it's stuck in `wfi` [here](https://github.com/oxidecomputer/hubris/blob/master/drv/stm32xx-i2c/src/lib.rs#L529)

Dumping the raw I2C peripheral space, we confirm that the `ISR` register only has `TXE` and `STOPF` flags set.

```
matt@igor ~ (sn5) $ h readmem 0x58001C00 0x30 -w
humility: attached to 0483:374e:003700313156501320323443 via ST-Link V3
                   \/        4        8        c
0x58001c00 | 001000d7 00010040 00000000 00000000 | ....@...........
0x58001c10 | 3060767f 000084c4 00000021 00000000 | .v`0....!.......
0x58001c20 | 00000000 00000020 0000001e 00000000 | .... ...........
```

In `I2C_CR2`, `nbytes` is 1, `SADD` is 0x40, which both look correct: we were beginning a read from the fan controller with address `0x20`, so shifted it's `0x40`

I was also running `watch -n10 pfexec humility ringbuf thermal` in a separate terminal, so maybe we interrupted something at exactly the wrong time?

Here's the dump:
[hubris.core.i2c_lockup.zip](https://github.com/oxidecomputer/hubris/files/9932368/hubris.core.i2c_lockup.zip)

cbiffle, 2022-11-04:
First pass analysis:

- The flag it's waiting on is `TXIS` (typo)
- The relationship between `TXIS` and `TXE` is not immediately obvious from the reference manual, except that `TXIS` seems to lag `TXE` in a multi-byte write. In this case it's never getting set, but `TXE` is set. It looks like the driver loop could be relying on `TXE` instead... but what's not clear to me is why this would ever work in that case, unless `TXIS` is somehow getting _cleared_ by our delay.
- I think we're configuring SMBus timeouts wrong, since they're not firing here. I think we need to configure TIMEOUTB when we're only setting TIMEOUTA.

mkeeter, 2022-11-04:
More data from the chat: here's the NVIC
```
matt@igor ~ (sn5) $ h readmem 0xE000E100
humility: attached to 0483:374e:003700313156501320323443 via ST-Link V3
             \/  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0xe000e100 | 00 00 00 80 25 00 00 20 00 02 04 80 01 00 00 00 | ....%.. ........
```
This shows IRQs 95/96 (I2C4 event + error) are enabled, as expected.

cbiffle, 2022-11-07:
One other thing we noticed on Friday: the driver never actually clears `STOPF`, so having `STOPF` set is a red herring. It's always set after the first transaction.

mkeeter, 2022-11-08:
One thing to remember for next time: we should dump the GPIOs to see whether the I2C lines are stuck low.

jgallagher, 2022-11-23:
I suspect it's not super useful, but in case that's wrong here's a humility dump from the sn5's SP: 
[20221123-fans-spinning-sn5.dump.tar.gz](https://github.com/oxidecomputer/hubris/files/10078016/20221123-fans-spinning-sn5.dump.tar.gz)

mkeeter, 2023-01-31:
Here's another case which looks like an I2C hang, captured on the `niles` Sidecar on 2023-01-31.

[hubris.core.1.zip](https://github.com/oxidecomputer/hubris/files/10549502/hubris.core.1.zip)

Unfortunately, we reset the chip before I remembered this issue, so didn't check the GPIO state.

cbiffle, 2023-04-27:
There've been a lot of robustness improvements to the I2C stack over the past six months, gonna close this as probably-obsolete.

closed 2023-04-27 by cbiffle