oxidecomputer/hubris #914: monorail task confused by Front I/O board?

jclulow, 2022-11-04:
I was trying to look at `humility monorail status` on the Sidecar Rev-A attached to **jeeves**, but the `monorail` task would reliably become confused and leave `hiffy` corked.  With some help from @arjenroodselaar I was able to chase this down a little, and I have put some notes together from our digging.

First, this reproduces on a build of the latest Hubris `master`, which at time of writing is 438eede6587ffb9327060f3440393c970cd0aefb.  I built a Sidecar SP image via:

```
$ cargo xtask dist app/sidecar/rev-a.toml
$ rsync -Paz target/sidecar-a/dist/default/build-sidecar-a.zip \
    jeeves:/data/local/jclulow/monorail/build-sidecar-a.438eede6587ffb9327060f3440393c970cd0aefb.zip
```

Flashing the archive shows the task is immediately upset (on **jeeves**):

```
$ ls -l /data/local/jclulow/monorail
-rw-rw-r--+  1 jclulow  staff      6.94M Nov  4 07:56 build-sidecar-a.438eede6587ffb9327060f3440393c970cd0aefb.zip
lrwxrwxrwx   1 jclulow  staff         60 Nov  4 07:57 build-sidecar-a.master.zip -> build-sidecar-a.438eede6587ffb9327060f3440393c970cd0aefb.zip

$ export HUMILITY_ARCHIVE=/data/local/jclulow/monorail/build-sidecar-a.master.zip
$ pfexec humility -t sidecar flash
...

$ pfexec humility -t sidecar tasks monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
system time = 2615
ID TASK                       GEN PRI STATE
13 monorail                     0   6 notif: bit31(T+10)

$ pfexec humility -t sidecar tasks monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
system time = 5597
ID TASK                       GEN PRI STATE
13 monorail                     4   6 wait: reply from net/gen0

$ pfexec humility -t sidecar tasks monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
system time = 8109
ID TASK                       GEN PRI STATE
13 monorail                     7   6 notif: bit31(T+12)
```

I held the task to get the panic message:

```
$ pfexec humility -t sidecar jefe -H monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
humility: successfully changed disposition for monorail

$ pfexec humility -t sidecar tasks monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
system time = 16782
ID TASK                       GEN PRI STATE
13 monorail                    15   6 FAULT: panicked at 'Could not initialize BSP: PhyPatchFailedCrc', task/monorail-server/src/main.rs:67:13 (was: ready)
```

I also collected the ring buffer:

```
$ pfexec humility -t sidecar ringbuf monorail
humility: WARNING: archive in environment variable overriding archive in environment file
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
humility: ring buffer task_monorail_server::__RINGBUF in monorail:
 NDX LINE      GEN    COUNT PAYLOAD
   0   50        1        1 ChipInit(0x80)
   1   66        1        1 BspInitFailed(PhyPatchFailedCrc)
humility: ring buffer task_monorail_server::bsp::__RINGBUF in monorail:
[... 127 lines cut ...]
 31  | SGMII   100M   2G5_23  6G_23   down | --       --         --
 32  | --      --     --      --      --   | --       --         --
 33  | --      --     --      --      --   | --       --         --
 34  | --      --     --      --      --   | --       --         --
 35  | --      --     --      --      --   | --       --         --
 36  | --      --     --      --      --   | --       --         --
 37  | --      --     --      --      --   | --       --         --
 38  | --      --     --      --      --   | --       --         --
 39  | --      --     --      --      --   | --       --         --
 40  | QSGMII  100M   1G_16   6G_14   up   | VSC8504  up        down
 41  | QSGMII  100M   1G_17   6G_14   up   | VSC8504  up        down
 42  | QSGMII  100M   1G_18   6G_14   up   | VSC8504  up        down
 43  | QSGMII  100M   1G_19   6G_14   up   | VSC8504  up        down
 44  | QSGMII  1G     1G_20   6G_15   down | --       --         --
 45  | QSGMII  1G     1G_21   6G_15   down | --       --         --
 46  | --      --     --      --      --   | --       --         --
 47  | --      --     --      --      --   | --       --         --
 48  | SGMII   100M   2G5_24  1G_0    up   | --       --         --
 49  | BASEKR  --     10G_0   10G_0   down | --       --         --
 50  | --      --     --      --      --   | --       --         --
 51  | SGMII   100M   2G5_27  10G_2   down | --       --         --
 52  | SGMII   100M   2G5_28  10G_3   down | --       --         --
```

I'm not sure what to do next!

Aaron-Hartwig, 2022-11-04:
I looked at this with @mkeeter this morning. Matt mentioned that this looked like it could be power related due to some prior debugging experience. Some attempts to talk to the PHY over the SMI manually yielded notably strange failures:
```
aaron@jeeves ~ $ pfexec humility -t sidecar hiffy -t ecp5_front_io -c Fpga.user_design_write_reg -adevice_index=1,addr=24,op=Write,value=2
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
Fpga.user_design_write_reg() => ()
aaron@jeeves ~ $ pfexec humility -t sidecar hiffy -t ecp5_front_io -c Fpga.user_design_write_reg -adevice_index=1,addr=25,op=Write,value=2
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
Fpga.user_design_write_reg() => ()
aaron@jeeves ~ $ pfexec humility -t sidecar hiffy -t ecp5_front_io -c Fpga.user_design_read -adevice_index=1,addr=19 --num=2
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
Fpga.user_design_read() => ()
Data: [0, 0]
```
Rather than yielding the data expected from the device, we read back `0x0000`. That is strange given that a non-responsive device expected to speak MDIO would result in a bus which floats high for the whole read, or `0xFFFF`. _suspicion of power intensifies_

Due to a rather invasive power change (see [MCN AE04](https://github.com/oxidecomputer/hardware-qsfp-x32/tree/main/RevA/MCN/AE04) for details), we cannot trust the FPGA with respect to what it knows about PHY power. The SMI interface between the FPGA and PHY has its signals pulled up to `V2P5_PHY_A2`:
[image]
This was useful to us during remote debugging because those pins go to the FPGA, so we could read their status. I made an adjustment to the QSFP FPGA image to keep the `MDIO` bidir pin in tri-state and added a register to our SPI-accessible interface, allowing us to see if it floated up or not, indicating that `V2P5_PHY_A2` was present. While we already had `MDINT_L` in that register interface, it is ambiguous if that signal would be low due to a power failure or the PHY driving it low.
The FPGA changes I made:
```diff
aaron@Aaron-Lenovo:~/Oxide/git/quartz/build$ gs
# On branch: main...origin/main  |  [*] => $e*
#
➤ Changes not staged for commit
#
#       modified: [1] ../hdl/MDIO/MDIO.bsv
#       modified: [2] ../hdl/boards/sidecar/qsfp_x32/VSC8562/VSC8562.bsv
#       modified: [3] ../hdl/boards/sidecar/qsfp_x32/VSC8562/vsc8562.rdl
#
aaron@Aaron-Lenovo:~/Oxide/git/quartz/build$ gd 1 2 3
diff --git a/hdl/MDIO/MDIO.bsv b/hdl/MDIO/MDIO.bsv
index d8a9809..5ddcd16 100644
--- a/hdl/MDIO/MDIO.bsv
+++ b/hdl/MDIO/MDIO.bsv
@@ -70,6 +70,7 @@ interface MDIO;
     interface Pins pins;
     interface Put#(Command) command;
     interface Get#(Bit#(16)) read_data;
+    interface Reg#(Bit#(1)) mdio_in_r;
 endinterface

 module mkMDIO #(Parameters parameters) (MDIO);
@@ -92,8 +93,8 @@ module mkMDIO #(Parameters parameters) (MDIO);
     // pin registers
     Reg#(Bit#(1)) mdc           <- mkReg(1);
     Reg#(Bit#(1)) mdint         <- mkReg(0);
-    Reg#(Bit#(1)) mdio_out      <- mkReg(1);
-    Reg#(Bit#(1)) mdio_out_en   <- mkReg(1);
+    Reg#(Bit#(1)) mdio_out      <- mkReg(0);
+    Reg#(Bit#(1)) mdio_out_en   <- mkReg(0);
     Reg#(Bit#(1)) mdio_in       <- mkReg(0);

     // Edge detection for MDC
@@ -139,8 +140,8 @@ module mkMDIO #(Parameters parameters) (MDIO);

     (* fire_when_enabled *)
     rule do_idle_bus(state == Idle);
-        mdio_out_en     <= 1;
-        mdio_out        <= 1;
+        mdio_out_en     <= 0;
+        mdio_out        <= 0;
     endrule

     // A future improvement to make would be to dynamically control preamble
@@ -249,6 +250,8 @@ module mkMDIO #(Parameters parameters) (MDIO);
     endinterface

     interface Get read_data = toGet(read_data_q);
+
+    interface mdio_in_r = mdio_in;
[... 27 lines cut ...]
+        } MDIO[2:2] = 0;
+
         field {
             desc = "1 for active interrupt (inversion of miim_phy_to_fpga_mdint_l pin)";
         } MDINT[1:1] = 0;
@@ -140,4 +144,4 @@ addrmap vsc8562 {
             desc = "Read = 0, Write = 1";
         } RW[0:0] = 0;
     } PHY_SMI_CTRL;
-};
\ No newline at end of file
+};
```
With suspicion mounting that `V2P5_PHY_A2` is not functioning, I flashed with my modified FPGA and read out the status register where my useful bits were held.
```
aaron@jeeves ~ $ pfexec humility -t sidecar hiffy -t ecp5_front_io -c Fpga.user_design_read_reg -a device_index=0x1,addr=0x12
humility: attached to 0483:374e:002900184741500520383733 via ST-Link V3
Fpga.user_design_read_reg() => 0x2
```
To decode this:
Bit 2 is MDIO, which is currently 0, meaning it has not floated high when we released the bus.
Bit 1 is MDINT, which is current 1. This is an inversion of MDINT_L, so we know it is low as well (and likely not because of the PHY).
Bit 0 is a don't care here.

While none of this is definitive, we strongly expect the rework done to the PHY power supplies (as mentioned above) to be broken in some way.

Aaron-Hartwig, 2022-11-07:
This was inspected by @refugeesus and the add-on board we kludged in to supply 5V bias for the LT3072 dev kit had it's output disabled via a pushbutton helpfully placed there. Toggling the button fixed things 🙃

closed 2022-11-07 by Aaron-Hartwig