oxidecomputer/hubris #945: Add `sidecar-sidechannel` task to support sidecar identity bootstrapping

jgallagher, 2022-11-17:
See RFD 329 and its discussion for full details. In the fullness of time, we will need a hubris task on sidecar that can talk to the host scrimlet over I2C to provide bootstrapping information: which sidecar is this (port config), what MAC addresses should be programmed, etc. For the short term, we're assuming only a single sidecar revision (rev-b), which will allow the host to bring up only the aux port (with a random MAC address) on which it can talk UDP to the sidecar SP to fetch the MAC addresses it should use.

We could smuggle "give me the MAC addresses" into either the MGS or transceivers UDP protocols and make `control-plane-agent` or `tranceivers` respond. It seems cleaner and more accurate to our long term plan to put this in a separate task (which can transition from UDP to I2C when appropriate), though.

mkeeter, 2023-02-13:
See https://github.com/oxidecomputer/transceiver-control/pull/58 and https://github.com/oxidecomputer/dendrite/issues/51

mkeeter, 2023-03-10:
Done in #1150

closed 2023-03-10 by mkeeter