oxidecomputer/hubris #947: Restart sequencer task if mainboard controller is reset

arjenroodselaar, 2022-11-18:
If some external event or entity resets the mainboard controller FPGA after the `sequencer` task has started it'll not continue to make forward progress. Consider the following log:
```
humility: ring buffer drv_sidecar_seq_server::__RINGBUF in sequencer:
 NDX LINE      GEN    COUNT PAYLOAD
   0  442        1        1 FpgaInit
   1  485        1        1 SkipLoadingBitstream
   2  495        1        1 MainboardControllerId(0x1de5bae)
   3  509        1        1 MainboardControllerChecksum(0x6461823f)
   4  541        1        1 MainboardControllerVersion(0x83)
   5  542        1        1 MainboardControllerSha(0x7b53e239)
   6  543        1        1 FpgaInitComplete
   7   25        1        1 LoadingClockConfiguration
   8  564        1        1 ClockConfigurationComplete
   9  568        1        1 FrontIOBoardPresent
  10   58        1        1 FrontIOControllerIdent { fpga_id: 0x0, ident: 0x1deaa55 }
  11   64        1        1 FrontIOControllerChecksum { fpga_id: 0x0, checksum: [ 0x72, 0xa5, 0x6b, 0xec ], expected: [ 0x72, 0xa5, 0x6b, 0xec ] }
  12   78        1        1 SkipLoadingFrontIOControllerBitstream { fpga_id: 0x0 }
  13   58        1        1 FrontIOControllerIdent { fpga_id: 0x1, ident: 0x1deaa55 }
  14   64        1        1 FrontIOControllerChecksum { fpga_id: 0x1, checksum: [ 0x72, 0xa5, 0x6b, 0xec ], expected: [ 0x72, 0xa5, 0x6b, 0xec ] }
  15   78        1        1 SkipLoadingFrontIOControllerBitstream { fpga_id: 0x1 }
  16  581        1        1 FrontIOVsc8562Ready
  17  121        1     5557 TofinoSequencerTick(Disabled, A2, None)
  18  402        1     5579 TofinoSequencerError(FpgaError)
  ```
  If a `TofinoSequencerError` is encountered the sequencer should check the device state of the mainboard controller and if it finds the device `AwaitingBitstream` it should restart itself. The FPGA will have its IOs floating, so nothing on the board works anyway and we may as well restart the task.

mkeeter, 2022-11-18:
We should also monitor the `ecp5_front_io` FPGA periodically and confirm that it's not in `AwaitingBitstream` (reconfiguring if necessary).

cbiffle, 2023-04-27:
Out of curiosity, what can asynchronously reset your controller FPGA?

cbiffle, 2023-07-10:
@arjenroodselaar can anything in Sidecar cause the controller FPGA to reset without the SP having done it? If the answer is no I'm going to close this.

arjenroodselaar, 2023-07-11:
AFAIK determine there shouldn't be anything cause a reset of the FPGA without the SP knowing about it. Yet this issue was the result of an instance where we found the sequencer running in it regular serving loop with the FPGA unconfigured. Which needless to say I did not expect to happen. A possible scenario could have been a power glitch causing a reset by the brown out detector, but we never figured this out. I also don't think we've seen it since.

Having said all that, it's not particularly difficult to add a check for the device state when encountering errors when reading from the FPGA and restarting the sequencer when it finds the FPGA in the wrong state, avoiding the system getting stuck.

If you want to close this that's fine, but I rather push it to the next milestone and have me look at this when I am back from vacation.