oxidecomputer/hubris #985: The SP should remember which host flash slot is active

jgallagher, 2022-12-05:
Currently, when `host-flash` starts, it [always chooses slot 0](https://github.com/oxidecomputer/hubris/blob/ee3787efcac4b1a3c90bdf11504aeb345bf9b1ce/drv/gimlet-hf-server/src/main.rs#L148), which is then [reported to the host](https://github.com/oxidecomputer/hubris/blob/ee3787efcac4b1a3c90bdf11504aeb345bf9b1ce/task/host-sp-comms/src/main.rs#L582-L599) when it asks. However, if the control plane has previously told the SP it should be booting flash slot 1, the SP should remember that choice if the `host-flash` task or the SP itself restarts. Currently changing the host flash slot only persists until `host-flash` restarts.

andrewjstone, 2022-12-14:
This information should be tracked and made available to the `stm32h7-update-server` similar to what is done in #1009 for the RoT.

mkeeter, 2023-01-27:
Should we also be saving the host startup options?

jgallagher, 2023-01-27:
Probably so, yeah.

mkeeter, 2023-01-30:
I had some good discussion in [#oxide-host-software here](https://matrix.to/#/!rHMNCYjVQiQsVlGkGP:oxide.computer/$HT2w6zmGIvB56wCc6NBmX8J2R-HW6_l-zolJeI5RjTQ?via=oxide.computer&via=unix.house&via=timecube.club), and have the core of a proposal.

The AMD boot process doesn't care about anything other than a few particular memory addresses in the SPIROM, which contain user-defined pointers to other data structures in the SPIROM.  None of these magic addresses are below 0x10000, so we could reserve the lowest sector (64 KiB) for the SP's use.

To remain robust against power loss at any point, we should use a ping-pong strategy with a monotonically increasing epoch value.  This is similar to the strategy in [RFD 311](https://rfd.shared.oxide.computer/rfd/0311), but uses an epoch instead of a checksum to select which slot to boot.  Note that unlike in RFD 311, where we ping-pong between slots within a single flash chip, here we're ping-ponging between the _two different flash ICs_!

Consider storing the following data structure at address 0 in the QSPI:
```rust
#[derive(AsBytes, FromBytes)]
#[repr(C)]
struct SpiHeader {
    empty: u32,   // must always be 0xFFFF_FFFF
    magic: u32,   // must always be 0x01de_abcd or something
    version: u64, // header version format, must always be 1 (for now)
    epoch: u64,   // monotonically increasing epoch counter
    active_slot: u32,
    boot_flags: u32,
    checksum: u32,
}
```

(address 0 should be 0xFFFF_FFFF because AMD firmware sometimes uses 0 as a marker for "empty"; 0xFFFF_FFFF won't confuse the boot ROM firmware into thinking that anything is present there)

We expect to change boot slots / flags _roughly_ as often as we reflash the SPIROM image itself, so there's no concern about flash wear.  The goal is **not** a general-purpose persistent storage mechanism; just enough to pick a boot slot and flags.  Indeed, this would be a particularly bad general-purpose persistent storage system, because you can only talk to it in A2!

Upon boot, we'd select the boot slot and boot flags by looking at each flash IC and reading the header from the one with the highest epoch.  This means that we could read the boot slot/flags from flash 0 (because it has the higher epoch), then decide to boot from flash 1 (because that's what the `SpiHeader` in flash 0 told us); this is fine!

Changing the active slot is a matter of
- Erase sector 0 in the QSPI with the lowest epoch
- Write sector 0 in that same QSPI with the new slot and an increased epoch

If at SP power-on, it notices that one QSPI has an invalid header, we can initialize it to a copy of the other header.  This ensures that we're always in a valid (redundant) state, even if we lost power or crashed mid-update.

If we go ahead with this plan, here's what it looks like:
- Update `amd-host-image-builder` to reserve sector 0
- Do not modify sector 0 when sending a QSPI image
    - Update Hubris to fail with an error if you try to do this
    - Update `humility qspi`, `faux-mgs`, and MGS proper to behave correctly
- Add Hubris APIs to persist data
- Update `humility qspi`, `faux-mgs`, and MGS to call those APIs after writing a host image

tagging @daym @jgallagher for opinions on this plan

--------------------------------------------------------------------------------

As an alternative, we could use [TLV-C](https://github.com/oxidecomputer/tlvc) here, e.g.
```rust
#[repr(C)]
struct SpiHeader {
    empty: u32,   // must always be 0xFFFF_FFFF
    magic: u32,   // must always be 0x01de_abcd or something
    tlvc: [u8], // flexible array
}
```

This would more closely match our RFD 311 EEPROM format, but tempts people into using this for general persistent storage, which we want to avoid.  Ideally, we'd never change the `SpiHeader` format!

jclulow, 2023-01-30:
I have two questions:

- sometimes, in the lab, we change `boot_flags` _a lot_ as part of tinkering with something; I know we're not super worried about wear, but what is the wear expectation for this cell?
- if one (but not two) of the ROMs is broken, we could still _update_ the functional one today; what impact will that have on the epoch strategy outlined above?

jclulow, 2023-01-30:
Perhaps we could have a way to change the `boot_flags` _but not update the non-volatile_ copy, for the lab?

mkeeter, 2023-01-30:
Yeah, I think we should be able to change `boot_flags` and `active_slot` in either RAM or ROM.  Maybe keep the current APIs for changing them in RAM, then add a different API to persist the change to ROM?

mkeeter, 2023-01-30:
> if one (but not two) of the ROMs is broken, we could still update the functional one today; what impact will that have on the epoch strategy outlined above?

This is an interesting question, and I'm not sure of the answer.  If we've only got one working QSPI flash, then by definition we can't be redundant, so we'd probably need some kind of special handling?

mkeeter, 2023-01-31:
Thinking about this a little more, I've got a proposed revision that improves redundancy and reduces erase cycle count.

Instead of storing a single `SpiHeader` at address 0, store _many_ `SpiHeader`s, e.g. every 128 bytes (it's currently 36 bytes, so this gives it room to expand).  This would let us fit 512 headers within sector 0.

On startup, we'd scan through both flash chips and pick out which header has the highest epoch, just like before.

When persisting data, we'd find the next available empty header location within sector 0, then write to it.  Only if the entire sector is full would we need to erase it.

This reduces the number of erases by a factor of 512, which lets us do one more thing: when persisting data, we should update **both** QSPI flash chips.  In other words, we want both flash chips to have the same highest-epoch entry at all times; if this isn't the case at task startup, the SP would fix up whichever flash has a lower epoch.

(this mimicks the `auxflash` task, which silently mirrors the auxiliary data if it detects that it's non-redundant)

These changes improve our performance if one flash chip dies:
- Having both flash chips always store a matching final entry means that if one chip dies, we haven't lost our boot flags / slot.  (If the boot slot is pointing to the dead flash chip, then we're going to have a bad time, but at least we haven't lost our persistent metadata)
- Using 512 slots means that if one flash chip dies and we need to update boot flags on the remaining chip, there's only a 1/512 chance that we'd pass through the Danger Zone of erasing the flash IC.

jclulow, 2023-01-31:
I like it!

andrewjstone, 2023-01-31:
This sounds good. I'm assuming based on the description, that both SPI ROMs contain identical images except for the header in failure cases, right?

One minor nit: I would call the `empty` field `reserved` or `amd_reserved` instead to be more explicit about its purpose.

Additionally, if we really wanted to save space, we could eliminate that field from all headers except for the first, although that may make the code a touch more complex.

jclulow, 2023-01-31:
> This sounds good. I'm assuming based on the description, that both SPI ROMs contain identical images except for the header in failure cases, right?

Not necessarily.  We haven't, as far as I know, got a solid policy on this yet.  The update engine might choose to keep an `N` image (current) and an `N-1` image (for rollback) during a phased roll-out of new OS images.

andrewjstone, 2023-01-31:
> Not necessarily. We haven't, as far as I know, got a solid policy on this yet. The update engine might choose to keep an `N` image (current) and an `N-1` image (for rollback) during a phased [...]

Ahh, I was assuming the matching because of the details about the matching header. But that doesn't matter because the active slot is listed. Thanks!

daym, 2023-02-01:
@mkeeter: Your newest proposal here looks good to me.

daym, 2023-02-01:
> One minor nit: I would call the `empty` field `reserved` or `amd_reserved` instead to be more explicit about its purpose.

I agree that the name could contain `amd` somewhere.

(What this field here is supposed to do is prevent AMD from trying to read our new struct as an embedded firmware directory. There's a magic number at the first 4 bytes of the directory that identifies it as a directory. Us leaving this magic number "empty" (0xffff_ffff) ensures that PSP firmware does not match our struct as a directory. The pointers from the amd main embedded firmware header to the respective directory sometimes contain 0 (instead of 0xffff_ffff as documented) to mean that that directory is not present. In that case, it could conceivably happen that the PSP firmware tries to read from there when trying to find the blobs)

mkeeter, 2023-02-14:
Done in #1145

closed 2023-02-14 by mkeeter