oxidecomputer/hubris #994: `hf` should at least respond to inquiries when it's upset

cbiffle, 2022-12-07:
Currently, if the hf driver fails to recognize the JEDEC ident info it reads from the QSPI host flash, it does this:

```rust
    if log2_capacity.is_none() {
        loop {
            // We are dead now.
            hl::sleep_for(1000);
        }
    }
```

This means that anyone who attempts to IPC the `hf` server will just wait forever, which is not great. It would probably be better to have the server come up in a known-degraded state, unable to (say) do programming but willing to respond to status inquiries or re-load the JEDEC info.

mkeeter, 2023-02-13:
Also, the `hf` task always muxes the flash to the SP when the task starts up.  @jgallagher points out in #1145 that if the task has died for some reason – but the rest of the system is in A0 – this could steal the flash away from the host!

nathanaelhuffman, 2023-03-17:
just to link this: https://github.com/oxidecomputer/hubris/issues/1172 this is the issue about expected muxing away

cbiffle, 2023-04-27:
> Also, the `hf` task always muxes the flash to the SP when the task starts up. @jgallagher points out in #1145 that if the task has died for some reason – but the rest of the system is in A0 – this could steal the flash away from the host!

Since the mux state is maintained by a GPIO, it should be straightforward to make `hf` respect its current state on task restart.

labbott, 2025-06-30:
Cosmo introduced the idea of a `FailServer` which returns an error each time. We now have that on gimlet after #2116. It's not quite " ...have the server come up in a known-degraded state, unable to (say) do programming but willing to respond to status inquiries or re-load the JEDEC info." but I think it's enough of an improvement over the existing behavior we can close out this issue and file a new one when we decide we want different behavior.