oxidecomputer/hubris #998: Upgrade `smoltcp`

mkeeter, 2022-12-07:
https://github.com/smoltcp-rs/smoltcp/issues/713 gives us good motivation to upgrade `smoltcp` from 0.8.0, once they put out a new release.

We'll also need to update [`gateway-messages`](https://github.com/oxidecomputer/management-gateway-service/blob/main/gateway-messages/Cargo.toml#L18) simultaneously, to avoid having two versions of `smoltcp` in the same build.

Right now, we're maintaining a `v0.8.0` branch [here](https://github.com/oxidecomputer/smoltcp/tree/v0.8.0), with that fix backported onto it.

jgallagher, 2023-02-13:
0.9.0 was released last week, and 0.9.1 a couple days later. dependabot kindly opened the [requisite gateway-messages PR](https://github.com/oxidecomputer/management-gateway-service/pull/59), so we can merge that whenever suitable.

cbiffle, 2023-04-27:
HELLO I AM FROM THE FUTURE
I AM PLEASED TO REPORT THAT MATT DID THIS IN THE YEAR TWO THOUSAND TWENTY THREE