oxidecomputer/hubris #1004: May want to move the SP/RoT image header

cbiffle, 2022-12-08:
Currently, we deposit an image header near the start of flash -- specifically, right after the vector table, since the CPU has opinions about where the vector table lives.

This has some drawbacks:

- It's useful to treat the image header as a "commit" point for writing an image, meaning that the update process may want to write the image header as the final sector written. Our current approach doesn't ensure that the image header is page-aligned, and requires any update logic to have intimate knowledge of the size of the vector table (which varies on every ARM implementation, meaning, the lpc55 and stm32h7 have very different header addresses).
- We may want to be able to conveniently extract the image header from a file on disk from generic tools. Currently this requires knowing which _processor variant_ the image is intended for. (Or heuristically scanning for things that look like a header). Such tools would be simplified if the header were in a predictable location.

I suggest that the _last page in flash_ is the right place. Our A:B partition layouts are fixed size for a part, so we can find the last page unambiguously. It's also easy to find the last block in a file on disk, for tooling. Finally, the last page in the flash address space is the last one written by a naive sequential update algorithm, removing special knowledge from updating tools.

We should be able to generate images with the header in the last sector as follows:
- Deduct one flash programming page from the FLASH `MEMORY` definition of the linker script for the part.
- Create an additional `MEMORY` definition containing just that one page, call it (say) `HEADER`
- Edit the existing linker script fragment that assigns image headers to right after the vector table, to instead put them in the new reserved area, by switching `>FLASH` to `>HEADER`.

This will produce an ELF file with a "hole" between the firmware and the header, which is probably what we want for right now -- in particular, this keeps SWD tools from flashing the unused space between the two, wasting programmer time during iteration. Eventually we want the hole to contain a predictable bit pattern for attestation reasons, and we probably want to do that with a post-processing tool either way; this is not a new concern from this bug, so I won't go into it further here.

@andrewjstone @lzrd @labbott @flihp

lzrd, 2022-12-08:
Is there any reason to have a compatibility phase where both the vector table header and the header trailer exists?
I'm fine with the answer "No".

lzrd, 2023-01-05:
An update that is destined to fail (wrong SKU, invalid epoch, linked for wrong bank, etc.) should be rejected by the target without erasing anything in the destination bank. That helps prevent excessive wear due to a misunderstanding with an upper layer that is retrying with an invalid image. That argues for showing a copy of the header to the target device at the start of the update process.
So, where ever the header lives, it probably should be an argument to the prepare update call.
This scheme introduces a dependency at the upper layers that they need to know where the header is. As long as this information is in block zero (the current case), we can be lazy. If we move it to some variable location (trailer), then it would be just the header structure, not the containing block that we would want available to the target at the beginning of the process.
Looking forward, including the header structure as an argument to prepare may be the best option.
probably its ultimate offset in the flash storage as well.

An integrity and signature check should be made as part of finish_image_update so that errors can be reported and the image can be left in a state such that stage0 will not select it.
Note, if we're changing something that affects how stage0 evaluates images and the updating (old) code does not understand the new method, then oops. So more thought needs to go into the design than just this paragraph.

lzrd, 2023-01-11:
Can we have the header before the vector table? There is nothing special about having the initial PC/SP at offset 0 followed by the vector table, right? We can have those elsewhere.
I'm thinking that the header should be at offset zero and an "activation block", a copy of the header, follow the image.
That way, we can get the entire image in flash, ask stage0 if it likes it (either reset and look at decision log or measurements, or some TBD hypocall to "runtime services").

cbiffle, 2023-01-11:
If we are booting through a bootloader, we can put the vector table at any location that is aligned to an implementation-defined boundary (often 1024 bytes). The bootloader would be responsible for setting VTOR correctly on control transfer. However, this layout of image wouldn't work for any machines _without_ bootloaders, which is currently "everything except the RoT," so I'm hesitant to do it since now we'd need different linker scripts for everything else.

Re: putting the header in twice to try and avoid coupling between layers -- think of this as an update file format analogous to ELF. The thing sending the software update is going to _have to_ have some awareness of its structure and characteristics of the target. If SKU validation etc. means the agent sending the update does this:

1. Clip out the last sector and send it for thumbs up/down on whether the update should proceed
2. Send all the sectors again

This seems fine to me.

I'd strongly encourage putting the header in the last sector as I suggested, since it produces an image layout compatible with all of our systems. Alternatives that I'm aware of either require clever out of order page writes (e.g. erase block 0 first but write page 0 last) or duplicating the header, which wastes precious flash.

lzrd, 2023-01-11:
Last sector it is.

"1. Clip out the last sector and send it for thumbs up/down on whether the update should proceed"
This would be an idempotent pre-flight check rather than the current update_server's prep_image_update() and write_one_block( ) calls which are destructive or a non-destructive dry-run version of those calls. We need stage0 runtime services to be able to give a high quality answer to "will it boot?".

Method for finding the image is simple since it is at the defined end of the bank. Stage0 and update_server have these
flash ranges compiled into them.

What's our recourse if we ever want to change the size of a bank?
We can say that we'll deal with that problem if it ever arises. We would have a new stage0 and it could be coded to know old and new header locations. Making it more complex now doesn't buy us anything (scanning for magic, etc.).

What is the installation image format?
It is simplest to have a copy of the entire bank contents including "filler" between the end of text/data and the header.
If we ever have R/W blocks or skipped blocks we would have to add meta information, likely to the header so that
it is available throughout the life cycle of the image.

If hypocalls are used to evaluate the newly flashed images before reset, then that would simplify some of the gymnastics needed to reliably transition to the newly written image.

In the RoT rollback case, we can write the lower version image to the alternate partition (e.g. B) and ask stage0 if it looks good (signature/ingetrity, SKU, and epoch are all acceptable). Given the policy of only booting the higher version image, we need to 'deactivate' the currently running image (A). SP has separate persistent state for that selection, but RoT does not.
It isn't safe for a running A partition to change any of itself for invalidation unless there is a block reserved for that purpose or the state is kept outside the bank.

The blocks that Cliff talked about using for a file system are available or we could take a block already part of the bank, probably move the header to last-1 to keep the image bytes contiguous and have the image activation state in the last block. Is there a better place?

cbiffle, 2023-04-27:
I think this is largely obsoleted by the combination of the image caboose, and the transition away from having a header between the hypovisor and Hubris image that gets validated. We do signature verification now, which is no longer susceptible to false positives due to partial or aborted writes.

closed 2023-04-27 by cbiffle