oxidecomputer/hubris #1069: transceivers-server shouldn't trash an entire request if one transceiver errors

Aaron-Hartwig, 2023-01-09:
Right now we [`Err` an entire request](https://github.com/oxidecomputer/hubris/blob/master/drv/transceivers-server/src/udp.rs#L480) if a single port has an error. This is a bad user experience and does not really give much actionable info back to the host.

We should put additional thought into this interaction and how we want it to work. I'd like to give all the good data that we can and then provide an error message for each port that failed, even if it would just be "Port X had an I2C error", which is better than the current state where we just error out completely:

```
BRM42220028-switch # xcvradm -i sidecar0 -f 0 identify
Error: Failed to identify transceiver modules

Caused by:
    0: Network protocol error
    1: Failure during read: FPGA reported an I2C error, module may not be present
```

Aaron-Hartwig, 2023-01-09:
@mkeeter: meant to CC you rather than assign, whoops. Brain fog. Just wanted to surface this as an improvement opportunity. @bnaecker please chime in if you have any thoughts here!

bnaecker, 2023-01-09:
I'm not sure what the right call here is. I think it'd be nice, though it definitely complicates the serialization / deserialization. In the meantime, you can avoid this kind of error with some new flags [I added](https://github.com/oxidecomputer/transceiver-control/pull/50) to `xcvradm` for addressing transceivers. You can now select based on on presence, management interface, and power-mode, as well as an index range (e.g., `0-5`).

The `identify` command requires a module to be present and in low-power, so you could do: `xcvradm -i sidecar0 -f 0 -t low-power identify` to accomplish what you want without errors.

Aaron-Hartwig, 2023-02-03:
> I think it'd be nice, though it definitely complicates the serialization / deserialization

True. And given the additional flags we have to address transceivers now, I don't think we need to do this. I think we will keep the current implementation where the SP attempts to give the host exactly what it asked for and will error if it can't talk to something it was told to talk to. However I would like to add another `Error` type to the messages crate. Something along the lines of a `ReadFailed`, but which can gave a port number rather than `HwError`. Thoughts on that @bnaecker?

bnaecker, 2023-02-03:
Adding more data to the error seems like a nice improvement! Does Hubris try to read each I2C buffer serially, failing on the first one? Or does it try to read all of them in parallel? I'm just wondering if we can send back a bitmask indicating all transceivers that failed, rather than a single index. I have no idea how complicated that is, and the latter would still be helpful!

Aaron-Hartwig, 2023-02-03:
Due to the memory layout here we are reading them all sequentially, so there really is no reason we couldn't keep a mask of failed transactions and return that with the error.

Thinking about that a bit more, it seems like we could potentially benefit from adding a "failed mask" to all message definitions since they can all be used on a single or many transceivers. If we failed, we could put junk data at its payload location. The host would know to disregard it because that module's bit would be set in the mask.

bnaecker, 2023-02-03:
Hmm, the idea of a failure mask is nice, although I'm not a huge fan of the junk data. It sounds easy to mess up, but I could be wrong that. Also, it's not obvious what the host-side does when it gets partially-good data. For example, if we're ultimately serving a request from the web console to list the transceiver power state, how do we show that? We'd need to make the response contain something like a result for each transceiver. I'm not sold on that. It might better to just fail the whole response, maybe giving the full failure mask. Also, giving the index of the _first_ failed transceiver would be better than what we have, too, if that's an easier first step.

Aaron-Hartwig, 2023-02-03:
>I'm not a huge fan of the junk data

I'm not either, but I'm not sure how else you get around that with fixed-length messages.

>It sounds easy to mess up, but I could be wrong that. Also, it's not obvious what the host-side does when it gets partially-good data.

It certainly would be, since we would have to use the failure mask to know when to disregard a certain range in the response buffer. If the SP were to change the order in which it interacts with the modules that doesn't feel like something we want to have to care about on the host.

>For example, if we're ultimately serving a request from the web console to list the transceiver power state, how do we show that?

Good question. What would we show them now?

> We'd need to make the response contain something like a result for each transceiver. I'm not sold on that.

Care to elaborate more on why you're not sold on that? The downsides are not immediately obvious to me. As it stands right now we just handle each result in sequence anyway. Ok: add data to the buffer, Err: send a response with an error.

>It might better to just fail the whole response, maybe giving the full failure mask. Also, giving the index of the first failed transceiver would be better than what we have, too, if that's an easier first step.

I agree that this seems like the most straightforward thing, although I'd like to hear more about that result-per-transceiver approach you mentioned. They both follow our strategy of having all the smarts in the host, not the SP. It is trivial to get the full mask instead of just the first failure.

bnaecker, 2023-02-04:
I should have been more clear. I'm thinking about this from the point of view of a consumer of Dendrite's HTTP API, most directly Nexus. I think the main reason the "result-per-transceiver" approach concerns me is that we need to return an HTTP status code to the client. Just like C's integer return code, this is a single value, and I'm not sure what to make it in this case. We definitely can make a result-per-transceiver, but then detecting an actual error requires parsing and inspecting the body, rather than checking if we got error-level HTTP status code. It's not impossible, but we'd probably want to think about it, and it also means we have to put that special logic into every client that consumes this data.

Now that I'm thinking about it, the other issue is that we just don't need it right now. The SDE only ever asks about _all_ transceivers, for things like presence, or a single one, when reading or writing. There's no way to communicate partial data to them.

I agree with all your other points. Fixed-length messages seem to imply we'd need junk data, and definitely we want to handle this on the host. I think let's go with a full error-mask for now, if that's easy, and if we decide having partial data is worth it we can do that as a follow-up. Does that sound reasonable?

Aaron-Hartwig, 2023-02-06:
Thinking about this from the perspective of a HTTP response helps crystalize the case for failing the whole transaction in the case of a single failure. Thank you for that!

The action I'm taking away from this discussion is to fail the whole request but send back a full error-mask with the error response. This is very low impact to existing code and is a clear improvement over the current state of things.

Aaron-Hartwig, 2023-02-10:
Here's an example of what I currently am toying with. I attempt to get `status` from 3 modules, 1 of which still has reset asserted and will not be able to respond.
```
aaron@niles ~ $ ./xcvradm -i axf2 -f 0 -t 0-2 status
FPGA Port Status
   0    0 PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT | POWER_GOOD
   0    1 PRESENT | ENABLED | RESET | LOW_POWER_MODE | POWER_GOOD
   0    2 PRESENT | ENABLED | LOW_POWER_MODE | INTERRUPT | POWER_GOOD
aaron@niles ~ $ ./xcvradm -i axf2 -f 0 -t 0-2 identify

Caused by:
    0: Network protocol error
    1: Failure during read: FPGA reported an I2C error, module may not be present. Logical mask for which ports saw a failure: 0x00000002
```

closed 2023-02-13 by Aaron-Hartwig