oxidecomputer/hubris #1071: minimize possibility of PUF identity key disclosure

flihp, 2023-01-09:
This includes:
* improved comments around the key code so folks understand the implications of granting access to it
* use of the IDXBLK_L register to prevent use of the key index used for this key
* more?

cbiffle, 2023-04-27:
@flihp, is the remaining work on this mostly to prevent disclosure during future maintenance of the code, or do we need more code to e.g. lock down the relevant key indices?

Marking this as MVP in case it's the latter, if it's the former we can move it.

flihp, 2023-05-01:
That's a good question. I lost a lot of state over the last few months so I'll triage this today and get you a better answer.

flihp, 2023-05-03:
Took a tour through the relevant code and it looks like this 'issue' was a sort of 'todo' list that never got updated before the persistent identity / PUF code was merged. The registers that block access to PUF resources are controlled through the interface [here](https://github.com/oxidecomputer/hubris/blob/7d0082ba02d8298b8c1ecc812ea654410ae0b645/lib/lpc55-puf/src/lib.rs#L218) and [here](https://github.com/oxidecomputer/hubris/blob/7d0082ba02d8298b8c1ecc812ea654410ae0b645/lib/lpc55-puf/src/lib.rs#L282). Both the `dice-self` and `dice-mfg` code paths [here](https://github.com/oxidecomputer/hubris/blob/master/lib/lpc55-rot-startup/src/dice.rs#L77) and [here](https://github.com/oxidecomputer/hubris/blob/7d0082ba02d8298b8c1ecc812ea654410ae0b645/lib/lpc55-rot-startup/src/dice_mfg_usart.rs#L225) block the index we use and then lock the relevant register.

There are always ways to improve this stuff, but AFAIK the items in this issue were addressed and the issue was unfortunately left open. Closing now.

closed 2023-05-03 by flihp