oxidecomputer/hubris #1073: RoT on Gimlet needs to respect the `GND_DETECT` line on the SP JTAG header

cbiffle, 2023-01-09:
The RoT on Gimlet has control over the SP's SWD port; this port is also routed to a header for use with an external programming dongle.

We ran a net to the RoT to detect presence of such a dongle, so that it can release the SWD pins to make sure that (1) the two drivers are not fighting and (2) SWD actually works. Currently the RoT is not respecting this line.

It probably should!

It looks like (@nathanaelhuffman to confirm) the board has resistors on the right nets to keep the drivers from destroying each other, so this is probably not a hardware damage issue.

@bcantrill @kc8apf

nathanaelhuffman, 2023-01-09:
Currently I'm concerned about the implementation on our current-gen boards.
[image]

I've opened issues to increase the series resistance on all the lines for the PVT spin, but above is the implementation on all our current hardware in-hand.

sidecar: https://github.com/oxidecomputer/hardware-sidecar/issues/755
gimlet: https://github.com/oxidecomputer/hardware-gimlet#1864
psc: https://github.com/oxidecomputer/hardware-psc/issues/317

isobering, 2023-01-10:
@cbiffle - for someone who wasn't involved in the original discussion, which ROT net is supposed to be detecting the presence of an SP dongle? Based on the net names alone, someone reading the Gimlet schematic would infer that both `ROT_TO_SP_SWDIO_MOSI_R` and `ROT_TO_SP_SWDIO_MISO_R` are intended to operate as data IO pins.

nathanaelhuffman, 2023-01-10:
@isobering I can help here (these shots from gimlet, but the other boards should have similar circuits):

RoT pin for SP jtag detection:
[image]

St-link or equiv dongle pulls that pin down when connected:
[image]

Note that the change we're making in hw is beefing up the series resistors to 300 Ohm such that if this detect fails we don't accidentally dead-short pins between RoT and dongle.

PR #1100 (2023-01-23): SP dumps from RoT represent inconsistent state
Fixes #1073 and #1099. Draft commit message:

```
This work fixes two bugs, #1073 and #1099. Beyond that, it also does
some minor refactoring to allow the (optional) use of `ringbuf` in the
kernel -- and adds a feature (optional and off by default) to Gimlet
to enable ring buffering of kernel events.
```
closed 2023-01-23 by bcantrill