oxidecomputer/hubris #1099: SP dumps from RoT represent inconsistent state

bcantrill, 2023-01-22:
In verifying the fix for #1073 and looking at SP dumps generated from the RoT, they appeared to represent inconsistent state.  This appeared to be explainable if register state were incorrect and/or if the target was not entirely halted when memory was read.  To explore this, the `--force-dump-agent` dump mechanism was modified to have a mode that stops the target from Humility (leaving it stopped), allowing for a dump to be taken via Humility, a dump to be subsequently manually initiated from the RoT, and then to be directly compared.  These results *should* be identical (the target did not run between the dump from Humility and the dump from the RoT), and indeed they were found to be similar -- but they were also disconcertingly different.

Here are [the two cores](https://github.com/oxidecomputer/hubris/files/10475017/cores.zip); from the dump taken from Humility *before* the dump was taken from the RoT:

```console
$ humility -d hubris.core.via-humility tasks
humility: attached to dump
system time = 3016854
ID TASK                       GEN PRI STATE    
 0 jefe                         0   0 recv, notif: bit0 bit1(T+46)
 1 net                         50   5 notif: bit31(T+34)
 2 sys                          0   1 recv
 3 spi4_driver                  0   3 recv
 4 spi2_driver                  0   3 recv
 5 i2c_driver                   0   3 recv
 6 spd                          0   2 notif: bit0(irq31/irq32)
 7 thermal                      0   5 recv, notif: bit0(T+348)
 8 power                        0   6 recv, notif: bit0(T+68)
 9 hiffy                        0   5 notif: bit31(T+32)
10 gimlet_seq                   0   4 recv, notif: bit0
11 hash_driver                  0   2 recv
12 hf                           0   3 recv
13 update_server                0   3 recv
14 sensor                       0   4 recv, notif: bit0(T+147)
15 host_sp_comms                0   7 recv, notif: bit0(irq82) bit1 bit2(T+142) bit3
16 udpecho                      0   6 notif: bit0
17 udpbroadcast                 0   6 notif: bit31(T+70)
18 udprpc                       0   6 notif: bit0
19 control_plane_agent          0   6 recv, notif: bit0 bit1(irq37) bit2
20 sprot                        0   4 recv
21 validate                     0   5 recv
22 vpd                          0   4 recv
23 user_leds                    0   2 recv
24 dump_agent                   0   4 recv
25 idle                         0   8 RUNNING
```

From the dump taken from the RoT from the halted SP and then extracted:

```console
$ humility -d hubris.core.via-rot tasks
humility: attached to dump
system time = 3016747
ID TASK                       GEN PRI STATE    
 0 jefe                         0   0 recv, notif: bit0 bit1(T+53)
 1 net                         49   5 ready
 2 sys                          0   1 recv
 3 spi4_driver                  0   3 recv
 4 spi2_driver                  0   3 recv
 5 i2c_driver                   0   3 recv
 6 spd                          0   2 notif: bit0(irq31/irq32)
 7 thermal                      0   5 recv, notif: bit0(T+455)
 8 power                        0   6 recv, notif: bit0(T+175)
 9 hiffy                        0   5 ready
10 gimlet_seq                   0   4 recv, notif: bit0
11 hash_driver                  0   2 recv
12 hf                           0   3 recv
13 update_server                0   3 recv
14 sensor                       0   4 recv, notif: bit0(T+254)
15 host_sp_comms                0   7 recv
16 udpecho                      0   6 notif: bit0
17 udpbroadcast                 0   6 notif: bit31(T+177)
18 udprpc                       0   6 notif: bit0
19 control_plane_agent          0   6 recv, notif: bit0 bit1(irq37) bit2
20 sprot                        0   4 recv
21 validate                     0   5 recv
22 vpd                          0   4 recv
23 user_leds                    0   2 recv
24 dump_agent                   0   4 recv
25 idle                         0   8 ready
[... 60 lines cut ...]
```

This represents a *deeply* impossible state: there should only ever be at most two different generation counts in a ring buffer.  This is clearly pointing to us getting wildly stale values on *some* reads.

It was helpful that `TICKS` is among these; a read of `TICKS` from the RoT via SWD *always* returns a result from the recent past (whereas via the STLink attached directly to the SP *always* returns the correct value).  But looking at how the AP and CSW being used by Humility (via probe-rs) and via the RoT (via SWD) shows no meaningful difference. (probe-rs was setting an additional bit in the CSW -- bit 29 -- but setting this bit did not change behavior.)

So they appear to be using the same sequence to read these values; **why is one of them wrong?!**

Fortunately, the STLink-V3SET has additional headers and allows for a logic analyzer to be easily applied.  Looking at the two different transactions via the logic analyzer revealed a very important difference; here is the CSW initiation from the SP:

[image]

And here is the CSW initiation from RoT:

[image]

There are several differences here, but the critical one is the `PROT` bits in bits 24 through 28; from the RoT, we are setting these to 0b00011, but from the STLink we are setting these to 0b01011.  Note that these are *not* the bits that probe-rs is setting -- and in particular, the STLink is setting this additional bit on its own!  Here is the definition of these bits in RM0433:

[image]

There is no further discussion of this bit -- it is merely called "Cacheable".  Modifying the RoT to set this bit in CSW results in it getting the correct value for `TICKS` -- and for everything else.  (That is, with this modification, the dumps that are expected to match do indeed match.)

The description for this bit is a tad misleading:  by "cacheable", the documentation would seem to imply that the load will allocate in the cache.  However, it empirically seems to mean that the load will *abide* by the cache -- and if the desired memory location is dirty in a cache, it will return that rather than the (stale) value in backing memory.  The STM32H7 has (by default) a writeback L1; if one were to ignore the L1 on reads and go straight to SRAM (which seems to be what having zero for this bit implies), one will get stale values.

And yes, [https://community.st.com/s/question/0D53W00000gesBbSAI/stm32h7xx-stlinkv2-cache-coherency-issue-when-debugging-from-keil-uvision5](others have hit this).

cbiffle, 2023-01-23:
looooooooool

Well, that explains that. If you'd like to try turning the SP's cache off, it's on line 145 of `drv/stm32h7-startup/src/lib.rs`. It sounds like that should cause the dumps to be coherent, at some cost in performance.

I'm concerned about `bufferable`. To me that implies the store buffer, a common feature on ARM processors at this end of the complexity curve. If we leave the bufferable bit clear, I wonder if it will bypass any store pending in the store buffer? That'd be annoying. The fact that the STLink is setting the cache bit suggests that a _lot_ of tools were leaving it blank and ST decided to take matters into their own hands; not setting the bufferable bit suggests that my fears here may be misplaced.

cbiffle, 2023-01-23:
It occurs to me ... the bug you linked is someone trying the STLink V2 with the H7 with the cache on. ST nudges people toward the V3 for the H7 -- it's required for the dual-core models for multidrop SWD, but not really required for the single-core models like we use.

Unless you're using proprietary debug tools.

From that post, it sounds like Keil uVision (proprietary, cannot be user-fixed) fails to set this bit and therefore always bypasses the data cache. This is probably why ST made the V3 patch this bit to 1 -- the proprietary JTAG interface is correcting for the behavior of the proprietary debug agent.

Sigh. This is how you wind up with Windows.

closed 2023-02-06 by bcantrill

cbiffle, 2023-02-17:
> It occurs to me ... the bug you linked is someone trying the STLink V2 with the H7 with the cache on. ST nudges people toward the V3 for the H7 -- it's required for the dual-core models for multidrop SWD, but not really required for the single-core models like we use.

This comment turned out to be prophetic, as @bcantrill has been chasing down a weird debug behavior with CMSIS-DAP probes on STM32H7...that appears to be probe-rs _also_ not setting this bit right.

bcantrill, 2023-02-17:
That comment turned out to be prophetic indeed! And ironic because one of my points of initial confusion was that we seemed to be setting the same bits from the RoT that probe-rs was setting -- and it was only when I got it on an analyzer that I could see that the STLink was silently setting additional bits.  This of course should have made clear that there would be a similar bug in the opposite direction, and I should have gone looking for the CMSIS-DAP equivalent -- but instead, it found us, as @andrewjstone started discovering very bizarre behavior when using an MCU Link against the H7 (that @cbiffle recognized as this issue). Filing an issue against probe-rs, and fixing in the branch that we use for Humility.