oxidecomputer/hubris #1108: Install My App - Doubt

Menniti, 2023-01-26:
Hello Oxide Team,

I do have 4 apps here, that each of then have been written on those 4 languages

1 C++ 
2 Golang
3 rust
4 python/cython

Q1 - Can I install all of then inside of Hubris? my application is latency sensitive (microseconds), so I believe I could be benefit with this kernel.
Q2 - if Q1, YES - Everything (hubris and my apps) needs a linux installed on machine?

All those tree has websockets, http apis, databases, queues and memory cache.

Thank you so much for the support

kc8apf, 2023-01-26:
1. Hubris isn't particularly designed for low-latency.  It is designed for fault isolation specifically in microcontroller applications.
2. Hubris can be built on Linux but does not require Linux for operation. 
3. Hubris has no websockets, http, database, or queues support built-in.  While those could potentially be created, Hubris is still focused on microcontrollers which do not map well to things like databases.

closed 2023-01-26 by kc8apf

Menniti, 2023-01-26:
Thankss